Qucs/qucs #330: review process reviewed

felix-salfelder, 2015-07-23:
the current review process has some issues. @guitorri has mentioned a WIP branch in #326. i call it "unstable", sounds more familiar.

from my experience, we should adopt some process that already works for other projects. here's my favourite
* everybody may open PRs, a PR is a request to merge code, unless stated otherwise
* an "issue" is a request to just discuss things.
* everybody may review PRs 
* a review can involve discussion about functionality and implementation details.
* a reviewer should read the code, filter out obvious shortcomings.
* a reviewer is entitled to request changes.
* a discussion that ends with at least two persons in agreement (typically the author and a person different from the author) is a positive review, if
* in addition some formal criteria are met (wasn't there a list somewhere...?)
* an official may now merge the PR into the `unstable` branch (see below), (s)he does not have to.

an `unstable` branch is the branch next to master, that will be merged into master by the release manager. unstable does not have to follow master (sometimes master needs a backported bugfix).
*maybe* there should be a `testing` branch on top of unstable, where one can prepare releases, while unstable is still being worked on.

that's all from me

felix-salfelder, 2015-07-24:
@ra3xdh if you don't agree with the unstable branch model, can we discuss this here instead of randomly mess with each others issues?

in3otd, 2015-07-25:
I see two main, not completely related, subjects here:
- the PR approval process
- the code development model 

(a PR approval process does not necessarily imply a certain code development model and viceversa)

Regarding the first, the process above is similar to the (informal) one we had so far. Looks fine, when things go well, everybody agrees and the sun shines. But, as we saw recently, the difficult part is how to handle any disagreement.
- what to do with the enthusiastic contributor who happily approves any PR, because ey wants to see some fresh air coming into the project at all cost? Will eir approval always count?
- how to handle that grumpy supporter, who has been around long enough that ey thinks that the long-standing bugs have now actually become features and should not be touched, so ey rejects most PRs. Will eir disapproval be taken into account?
- what if, after an unstable branch has happily grown with the support of a couple of people, a developer shows up saying, "sorry, I was too busy with real life to take a look at this before; I have just gone through the code and I think it's really a bunch of crap". 
- and countless similar situations

(yes, any resemblance to real events and/or contributors, past, present or future, is purely coincidental)
(no, I do not have a general solution for the above situations)

Enough for an inconclusive first part.

A couple of maybe more constructive questions:
- should a PR be "a request to merge code" or "a request to merge a **feature**" ? It is not always possible to say whether a PR is good if it is not clear what the end goal of the code is.
- do you really mean there should be just **one** unstable branch, where the code from all (unrelated) PRs go??
- I would like to add that besides reading the code, a reviewer should be expected to also compile and test the code in its own environment, especially before giving the ok to a PR.

Of course we are not the first to think about which development model fits a project better: one quite popular seems to be [this one] (http://nvie.com/posts/a-successful-git-branching-model/) (I think guitorri also proposed this some time ago). Seems quite good to me, even if it requires a bit more work to maintain the different "main" branches.

felix-salfelder, 2015-07-25:
thanks for your thoughts. indeed i mix up "approval" and "development". this is what i got used to as a working concept, given there is no evil collaborator who merges in code without consent. discussion will often help in the special cases you list, ymmv.

> should a PR be "a request to merge code" or "a request to merge a feature" ? 

you can always argue that a code change is a feature and vice versa. whether or not you get through with this, depends on whether others want to merge your PR. i.e. if we decide to only merge "features", would you still merge a PR that changes header guards? if we decided to only merge code, then, are header guards really "code"?

most importantly, a PR should not break anything. the more obvious this is, the easier it can be reviewed/merged. some things can be reviewed by just reading the code. if a PR changes whitespace only, it will not make sense to even compile before a sane and valid merge. some others need excessive interactive testing, in addition to a careful read of the change set.

so arguably, you ask the wrong question. not what you call your feature, but the *number* of features is significant. a PR with more than one feature or even a *buch of* independent features is really and has been the issue. sometimes such a monster-PR has the (intended or unintended) side effect that some other work is stalled, particularly once the mere existance of a PR is used as a pretext against other PRs. multiple non-conflicting PRs do not have these problems, and in addition, there's one thread to discuss each of the feature independently.

as you have guessed: i'd be in favor of having at most one feature per PR, be this an approval criterion or a development model.

> do you really mean there should be just one unstable branch

yes. FWIW: i like the link. coincidentally, the proposed (unstable,testing,master) branches correspond more or less exactly to the branches (develop,master,release) in the figure in
http://nvie.com/posts/a-successful-git-branching-model/. and there is only one "develop" branch.

in3otd, 2015-07-27:
strangely enough, I tend to leave (in part) the definition of "feature" to the (intentions of the) author of the PR. If the intention is to just change header guards, once this is done I will merge it (ok, assuming it is useful, properly done, etc., that's not the point here). If the PR initially is about removing leading/trailing underscores and the author says "maybe we should also change macro names", I will probably wait for another commit with these changes also before merging.
For this simple example it does not make much difference, but in general I prefer to see a more substantial amount of changes in a PR, I think it helps in understanding how the code (feature?) fits in the general project development.
If the PR is about a long-term development, I will tend to keep it separate from the rest of the code. What if the author disappears without notice in the middle of the development (is his/her/its right -  no questions asked)? Will we keep in the main code base the unfinished work? Or manually carve out the unused code? 
What if some widely used class need a change of name/definition because halfway in the development the author has a better idea about the implementation? Will we record permanently in the history also these changes, which will very likely be unimportant once the "feature" is finished?

Regarding the development model in the link, I think the main difference in our points of view is that form me most of the action is in the "feature branches" and develop should be mainly the result of merging the various "feature branches", while I think you see develop as the place where all the PR are directly merged.
I see the feature branches as the place where the main author(s) pushes its code, fixes it, maybe squashes some stuff together at some point and maybe even rewrite the branch history, all independently from the develop branch.
Reusing the example above, yes, a header guards change can be merged directly in develop but something more substantial will need to have a dedicated branch, IMHO.
This is also how I understand the [image in the link] (http://nvie.com/img/git-model@2x.png) and the entire article.

felix-salfelder, 2015-07-27:
> Or manually carve out the unused code? 

this is a good point. in some sense, a PR shouldnt be allowed to add dead code.
there are projects where you cannot add branches that are not reached by the included tests, unless e.g. you add an `untested()` call at the begiining of that branch. with this you could easiliy identify potential dead code and refuse to merge it.

(afaik no such test and reachability rules exist for QUCS, would you like to see them added?)

> What if some widely used class need a change of name/definition

yes, this happens. and it's slightly ugly. but think positive. here the codebase gets improved by a change as harmless as it can be. *and* nobody could think of it a few days/commits earlier. so it's certainly worth the trivial review.

> What if the author disappears without notice in the middle of the development 

well, if somebody leaves behind one complete feature to be merged, (s)he will have contributed more than somebody who leaves behind random incomplete work on two features. sure, ymmv.
still, nobody is forced to merge/review either case.

> I think you see develop as the place where all the PR are directly merged.

i assume "merging in features" in the article is what corresponds to PRs. yes, you are right. i can not find a second branch in the article that features are merged into. other than that, i do not understand what you identify as said difference. there's not even a contradiction

code that does not get reviewed does not get in (hopefully). code does not improve by just sitting there. if it contains intricate bugs, nobody will notice. limitiations will not be identified unless work continues.

> Reusing the example above, yes, a header guards change can be merged directly in develop but something more substantial will need to have a dedicated branch, IMHO.

even the header guards example needs a dedicated branch. the header guards PR somehow corresponds to this branch. what is "substantial" supposed to mean, and what is special about the (maybe even more) dedicated branch? still cannot find such a branch in the article.

thanks

guitorri, 2015-10-02:
I like the `master`, `develop`, `hotfix` and `release-[x.x.x]` branching approach.
Anything else are `feature-branches` either living on the main repository or in a PR from somebody. 
If we decide for it, we have to find the best moment to adopt it.

A few other details come to mind:
- Use of namespaces for work done on the main repository.  
  - Perhaps it makes sense to keep all the 'core' development inside the repository. Keeping track of work (perhaps already) done in forks might become a problem. 
  - People who wish to contribute more regularly would get a `[user initials]/` namespace (or something similar) in the repository. It might also simplify a bit the PR on top of PR recursion in case two or more developers are doing some work together.

- Every merge happens thru PR. It is already happening, but just formalize it somewhere.

- Use the milestones tags to prioritize the work. It helps us focus on the release cycle and blockers.

- A PR is a request for merge, unless stated otherwise. The sender should mark it as WIP or the reviewer should mark it as incomplete. It should be clear from the title or labels that the PR is not ready.

felix-salfelder, 2015-10-02:
this sounds very good. let me add two remarks to the grey area.

> Perhaps it makes sense to keep all the 'core' development inside the repository. Keeping track of work (perhaps already) done in forks might become a problem.

well, we have those PRs with WIP status. this is github specific, and will be lost when moving away from github.... i must say i hate this lock-in. but then, the workflow is pretty efficient.

> It might also simplify a bit the PR on top of PR recursion in case two or more developers are doing some work together.

i am against merging PRs into each other, every PR should target one of the "official" branches, and constitute an improvement (not necessarily a feature!). PR clusters with authors refusing (or lacking time) to untangle/split are getting problematic quickly. this does not happen when stacking PRs, which is not dramatically difficult.

as an example, we could have one PR "support .va files in libraries" and one PR "add a .va library file", and a next "include verilog models into verilog netlist output". none of those would really interfere with other people's work (candidates for `develop`!). currently, we have one huge project that tries something similar. it is implemented in a questionable way, is not open for discussion and is based on another otherwise independent PR. after a merge, any attempt to clean up (read: improve) will certainly break a lot of "features" currently built on sand...

guitorri, 2016-02-14:
I think we should adopt the so called git flow branching model [1]. 

If needed, the workflow is supported by scripts [2]. It is even integrated in some graphical front end tools for git [3]. 

I think it is clear that GitHub-flow we currently use is standing on our way for releases and parallel development. See [4] for a comparison.

There are custom variants of git flow such as [5] and [6]. Note that the long-lived branches have other names.

We should not attempt to craft another branching model, unless we see the need for that.

I will soon create a `develop` branch (for merging code already reviewed, cutting edge branch, content for future release, basically it is our current `master`) and a `release-0.0.19` (bug-fixing related to 0.0.19). The `master` branch will *gradually* become the branch where tags for final released versions end up. Please read [1] carefully. 

I said *gradually* because we have a already several PRs addressed to `master`. There are useful discussions on those PRs. If not for the discussions I would ask people to close and re-send the PR against `develop`. 

While we do the transition it will be fine to merge `master` into `develop` as needed. After this transition period (hopefully done by 0.0.20), `feature` branches shall be merged (via PR) against `develop`. A PR can arrive from personal forks or from the `origin` repository (in case of collaborative branches or branches that need early CI integration).

How does that sound?

[1] http://nvie.com/posts/a-successful-git-branching-model/
[2] https://github.com/nvie/gitflow
[3] http://lucamezzalira.com/2014/03/10/git-flow-vs-github-flow/
[4] https://www.atlassian.com/git/tutorials/comparing-workflows
[5] https://www.bro.org/development/howtos/process.html
[6] https://wiki.qt.io/Branch_Guidelines

in3otd, 2016-02-15:
we can also merge the outstanding PRs *manually* to the `develop` branch, even if these were originally addressed to `master`. It's a (little) bit more work for us, but I think it will be clearer this way. 
Apparently you can [cross-ref the open PR] (http://stackoverflow.com/questions/9135913/merge-pull-request-to-a-different-branch-than-default-in-github)  (and maybe even close?) in the merge commit to `develop` so there will be a trace in the PR discussion of where the PR ended up.

felix-salfelder, 2016-02-15:
> It's a (little) bit more work for us

i always type in the 3 commands for a merge, rather than trusting some magic button. i'd expect that github will notice the merge, even if it goes to "develop", but i have not tried.

apart from that, i am be happy with [1], including all the nomenclature and choice of names.

in3otd, 2016-02-15:
> i always type in the 3 commands for a merge

yep, often I get lazy in the last step and just press the green button instead of pushing from the command line. Since the PR is already pulled locally for review and test (except for trivial changes), pushing to `develop` instead of `master` should be just a few more characters to type...

I was looking for a way to avoid accidentally pushing to `master` instead of `develop`, since I will expect (in the long term) that this will automatically trigger a release and so upload the release artifacts to GitHub/SourceForge from Travis and AppVeyor, but I didn't find much, except some limited stuff [here] (https://help.github.com/articles/about-protected-branches/) (still, protect `master` as described there seems a sensible thing to do)

felix-salfelder, 2016-02-15:
> still, protect master as described there seems a sensible thing to do

makes sense. I failed to figure this out with github. `:(`...

yet, we could work around this by declaring the master branch on sourceforge authoritative. and lock it down reasonably.

guitorri, 2016-02-15:
The branches `develop` and `release-0.0.19` are up. 

If you merge locally, don't forget the `--no-ff`.
Please help steering new PRs to `develop`. Maybe we should make this the default branch on GitHub.
Please tag things according to the release milestone.

in3otd, 2016-09-05:
clearly at present we do not have enough active developers to make the usual PR/review/merge process work properly. The PR opened do not get reviewed in a reasonable time, if at all.
What shall we do? Given the current state of things I'll be ok if PR are self-merged in `develop` after, let's say, one week of being opened. At least Qucs will go somewhere instead of stagnating.
For PRs to `release-0.0.19` I'll tend to be more conservative as the branch is (in theory) closer to become a release so will undergo less general testing. But again, what shall we do here?

felix-salfelder, 2016-09-05:
> What shall we do?

that's simple: modularize.

adding or changing extensions could be completely up to the author(s) and go into some space reserved for such contributions (`contrib`) even without review.

i am aware that we cannot currently make any use of such extensions in master/develop or in release-0.0.19. but we have never been closer (look at the PRs). i don't think it's the right time to give up *now*.

> let's say, [merge what you want] one week of being opened

at the current pace, one week is really short. how about one month?

i'd favor a more practical approach that does not bypass review completely. just go find someone who (after one month or so) puts "hey I have seen your commits, i don't have time for a thorough review, but travis is fine. also, i think it makes sense, and nobody has objected until now -- feel free to merge".

about whether reviewing "rules" should depend on the targeted branch, i am not sure.

felix-salfelder, 2020-09-26:
of little relevance. closing.

closed 2020-09-26 by felix-salfelder