Qucs/qucs #333: QucsGUI - `master` crash on specific schematic in command line

ra3xdh, 2015-07-25:
Bug presents in `master` branch

Steps to reproduce
1. Qucs-GUI, CMake-build , `master` branch, Linux platform
2. Run Qucs-GUI with project from `qucs-test` suite:
~~~
/path_to_builddir/qucs -n -i testsuite/AC_bandpass_prj/bandpass.sch -o testsuite/AC_bandpass_prj/test_bandpass.txt
~~~
3. Qucs crashes. Here is a GDB backtrace:
~~~
(gdb) bt
#0  0x00000000005dff6e in QHash<QString, QString>::value (this=0x30, akey=...) at /usr/lib64/qt/include/QtCore/qhash.h:609
#1  0x00000000005ea0d9 in Subcircuit::getSubcircuitFile (this=0xe3f8b0) at /home/vvk/qucs/qucs/qucs/components/subcircuit.cpp:351
#2  0x00000000005e79ba in Subcircuit::createSymbol (this=0xe3f8b0) at /home/vvk/qucs/qucs/qucs/components/subcircuit.cpp:79
#3  0x0000000000701bad in MultiViewComponent::recreate (this=0xe3f8b0, Doc=0x0) at /home/vvk/qucs/qucs/qucs/components/component.cpp:1369
#4  0x000000000070642d in getComponentFromName (Line=..., p=0xe38dc0) at /home/vvk/qucs/qucs/qucs/components/component.cpp:1709
#5  0x0000000000508acb in Schematic::loadComponents (this=0xe38dc0, stream=0x7fffffffda00, List=0x0) at /home/vvk/qucs/qucs/qucs/schematic_file.cpp:642
#6  0x000000000050ac88 in Schematic::loadDocument (this=0xe38dc0) at /home/vvk/qucs/qucs/qucs/schematic_file.cpp:894
#7  0x00000000004a1784 in openSchematic (schematic=...) at /home/vvk/qucs/qucs/qucs/main.cpp:272
#8  0x00000000004a1885 in doNetlist (schematic=..., netlist=...) at /home/vvk/qucs/qucs/qucs/main.cpp:282
#9  0x00000000004a76ed in main (argc=6, argv=0x7fffffffe358) at /home/vvk/qucs/qucs/qucs/main.cpp:994
~~~

There is a problem with `QucsMain->schNameHash` calculation. From GUI this circuit could be opened normally.

guitorri, 2015-07-25:
There are a couple of issues.

- the `QucsMain->schNameHash.value(baseName)` does not exist from the command line. QucsMain is not launched yet. A simple `if (QucsMain)` guard to skip the hash access guard allow the schematic to be printed...

- then comes the second issue, the subcircuits are not loaded. The is the problem that `Subcircuit::getSubcircuitFile()` is failing return its absolute (or even relative) path. So it cannot be loaded.

Workaround: If the command is run from inside the project directory, which contains the parent schematic and the subcircuit, things seem to work fine. Both the netlist and print functions.

```
cd testsuite/AC_bandpass_prj/
/path_to_builddir/qucs -n -i bandpass.sch -o test_bandpass.txt
/path_to_builddir/qucs -p -i bandpass.sch -o bandpass.pdf
```

guitorri, 2015-09-13:
With PR #351 following should work 

```
qucs -n -i t foo.sch -o netlist.txt
``` 

Or running all the qucs-test examples:

```
python run.py --prefix /home/user/local/qucs --qucs
```

I did not test schematics with components including Verilog/VHDL textual models. 
QUCSDIR might need to be set to a installed version of Qucs in order to use qucsconv (for spicefile components) and library components.

closed 2015-09-14 by guitorri