Qucs/qucs #335: Qucs-help fail to find documentation.

guitorri, 2015-07-25:
Issue reported by Walter Woods:

> Running qucshelp from the terminal or by pressing F1 in the program both causes an immediate 
> crash.  For what it's worth, the terminal output when I launch it that way is:
```
Warning: HtmlDataFetcher::fetchLinksToFiles() : File open error
Warning: HtmlDataFetcher::fetchChapterTexts() : Can't open file NONE/share/qucs/docs/en/index.html
Fatal: ASSERT failure in QList<T>::operator[]: "index out of range", file /usr/local/include/QtCore/qlist.h, line 477
/usr/local/bin/qucshelp: line 2: 12479 Abort trap: 6           /usr/local/bin/qucshelp.app/Contents/MacOS/qucshelp $*

```

wwoods, 2015-07-26:
I have files in /usr/local/share/qucs/docs/en/.

No --prefix was specified with configure.

Running qucshelp like this works:

    QUCSDIR=/usr/local qucshelp

in3otd, 2015-09-18:
...strange, can you post (or use [gist] (https://gist.github.com/)) the `config.log` file that should be in the build directory?

felix-salfelder, 2015-09-29:
maybe also paste config.h, mine looks like this.

    /* Where the documentation files go. */
    #define DOCDIR "NONE/share/qucs/docs/"

we should not hardcode DOCDIR (and others?). it should be composed from `${prefix}`. also we should stick to the standard. see Makefile, there we have `docdir = ${datarootdir}/doc/${PACKAGE_TARNAME}`

in3otd, 2015-10-01:
strange, you too have `NONE` in `DOCDIR`, never seen that before

my limited understanding of how all the autconf stuff works will say that this should not happen, the path above is derived from `$prefix` which in the end should always get a value other than `NONE`.

> we should not hardcode DOCDIR (and others?). it should be composed from `${prefix}`

currently it should be already composed from `${prefix}`, https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L780-L781 , no ?

felix-salfelder, 2015-10-01:
`$prefix` in configure.ac has a different meaning. namely the string passed through the `--prefix` option. omitting `--prefix` yields `$prefix=NONE`.

in one of my other configure.ac's, i did this.

    # stupid hack (needed?)
    if test "x${prefix}" = xNONE; then
      AC_DEFINE_UNQUOTED( PREFIX, "${ac_default_prefix}", "Prefix")
    else
      AC_DEFINE_UNQUOTED( PREFIX, "${prefix}", "Prefix")
    fi

hth

felix-salfelder, 2015-10-01:
> currently [DOCDIR] it should be already composed from ${prefix}

it is hardcoded in `config.h`, this way, the dependency from ${prefix} is lost.

hmm, i cant think of a proper way to do this right now. maybe we should steal some other projects solution. for now, working around the `NONE` case should be sufficient.

guitorri, 2016-02-05:
On Windows, if QUSDIR is not set, it will crash on start.

guitorri, 2016-02-06:
I guess this is related to #451.
To have things relocatable, we should not set them at build time.
Perhaps one solution is to let Qucs register its location in the QSettings during the first launch. Fron that on, all other tools can pick up the location and find files relative from that (with some assumptions).
Note that the build and install layout is different for qucs-help and qucs-lib... i mean, executables and loadables (html, libs) will be in different places relative to each other.

in3otd, 2016-02-10:
a half-fix could be to use in the tools the same code as in the [Qucs application] (https://github.com/Qucs/qucs/blob/master/qucs/qucs/main.cpp#L652-L686) . Every tool looks for the language/lib/help/etc. files in a path relative to its own location. Of course this won't work for the build layout, but should at least fix the issues with the Windows `.zip` distribution, since there we are sure we have the install layout (or not ?)

guitorri, 2016-02-16:
Fix merged into `release-0.0.19` branch (879c4cfe9772b60b954ab095b6b9f6e3c65761e3)

closed 2016-02-16 by guitorri