Qucs/qucs #342: Using a substrate with er=1 produces invalid results (NaN) in S-Parameter simulation

StefanD986, 2015-08-23:
When using a substrate with the relative permittivity "er" set to 1.0, the S-Parameter Simulation yields only results with "NaN". See below for an example schematic.
I had a look at the equations of the Models and my guess is: With EpsilonR set to 1 it equals Epislon_R_eff. In some equations for the Microstrip models is "EpsilonR - EpsilinR_eff" in the denominator, and then probably causes the NaN result.

My configuration is: 
Qucs 0.0.18 (installed via guitorri/homebrew-tap)
Mac OSX 10.10.5

```
<Qucs Schematic 0.0.18>
<Properties>
  <View=0,0,800,800,1,0,0>
  <Grid=10,10,1>
  <DataSet=epsilonR_bug_example.dat>
  <DataDisplay=epsilonR_bug_example.dpl>
  <OpenDisplay=1>
  <Script=epsilonR_bug_example.m>
  <RunScript=0>
  <showFrame=0>
  <FrameText0=Title>
  <FrameText1=Drawn By:>
  <FrameText2=Date:>
  <FrameText3=Revision:>
</Properties>
<Symbol>
</Symbol>
<Components>
  <Pac P1 1 240 220 18 -26 0 1 "1" 1 "50 Ohm" 1 "0 dBm" 0 "1 GHz" 0 "26.85" 0>
  <MLIN MS1 1 410 170 -26 15 0 0 "Subst1" 1 "1 mm" 1 "10 mm" 1 "Hammerstad" 0 "Kirschning" 0 "26.85" 0>
  <GND * 1 240 250 0 0 0 0>
  <Pac P2 1 510 210 18 -26 0 1 "2" 1 "50 Ohm" 1 "0 dBm" 0 "1 GHz" 0 "26.85" 0>
  <GND * 1 510 240 0 0 0 0>
  <.SP SP1 1 300 480 0 51 0 0 "lin" 1 "1 GHz" 1 "10 GHz" 1 "19" 1 "no" 0 "1" 0 "2" 0 "no" 0 "no" 0>
  <SUBST Subst1 1 620 260 -30 24 0 0 "1" 1 "1 mm" 1 "35 um" 1 "2e-4" 1 "0.022e-6" 1 "0.15e-6" 1>
</Components>
<Wires>
  <240 170 380 170 "" 0 0 0 "">
  <240 170 240 190 "" 0 0 0 "">
  <510 170 510 180 "" 0 0 0 "">
  <440 170 510 170 "" 0 0 0 "">
</Wires>
<Diagrams>
</Diagrams>
<Paintings>
</Paintings>
```

in3otd, 2015-08-23:
thanks for the bug report; as you saw, in the microstrip code there are some expressions using the ratio `(ErEff - 1)/(er - 1)`, which turns out to be `0/0` when `er` is exactly equal to one. Will try to see if it can be rewritten differently or, at worse, checking for the special case `er=1`.
In the meantime, a workaround is to use an `er` value slightly above zero, like `1.001`...

StefanD986, 2015-08-23:
I guess an ```er=1``` occurs only theoretically and I found it more or less by accident. (the only thing that I can imagine might be Aerogel substrates?)
So its probably not worth the hassle to try to rewrite the equations. But what might be a good idea to help anyone in the same situation to identify the cause of the ```NaN```results, is to throw a warning in the simulation log if ```er==1```.

in3otd, 2015-08-24:
AFAIU, only one equation has issues with `er` equal to 1 (the one for the dielectric losses), I had the impression the equation could be rewritten differently or at least I think I usually saw it in a different form.
FYI, I saw some simple structures realized compeltely in air for high-power applications (couplers, etc.), where the lines are simply suspended between connectors (these were more stripline-like than microstrip, actually)