Qucs/qucs #345: Document the receiver() function

endolith, 2015-08-28:
Trying to calculate total noise over a given bandwidth, I found this post "[noise root mean square with logarithmic AC-analysis?][1]":

>> i was wondering if there is an easy way to calculate the root mean
>> square noise value from the spectral density if the simulation step is
>> non linear (logarithmic)?

> are you asking for noise in a bandwidth, right?

> For this purpose we have a receiver() equation, see english help on
math functions...

but I don't see this function anywhere in the documentation.

By [searching through the source code][2], I've found that there *is* a `receiver()` function, but it doesn't seem to have anything to do with integrating noise spectra:

    /* The function computes a EMI receiver spectrum based on the given
    waveform in the time domain. The number of points in the waveform
    is required to be a power of two. Also the samples are supposed
    to be equidistant. */
    vector * emi::receiver (nr_double_t * ida, nr_double_t duration, int ilength) {

  [1]: http://sourceforge.net/p/qucs/mailman/message/31413820/
  [2]: https://github.com/Qucs/qucs/search?q=receiver

guitorri, 2015-09-17:
The function does takes a timed-domain waveform, does FFT, sums up the contributions and adds noise floor to the calculations. 
To access the receiver() function you need to use an Equation block. Indeed the function is not documented.  
The [function signature](https://github.com/Qucs/qucs/blob/8c2dd9e909f0554ee0d1e4a305e54ad68191ed8e/qucs-core/src/evaluate.cpp#L4739-L4747) in a equation should be something like 
`emi = receiver(data, duration, [FFT length])`.

If you figure it out please let us know so we can document it...

endolith, 2015-10-07:
Is thermal noise simulated during a time-domain simulation?  I thought it was only simulated during AC sweeps with noise enabled.  I just want to calculate total noise (not spectral density) from AC sweeps.