Qucs/qucs #347: Wrong result in impedance calculation for rectangular Waveguide with dielectric medium

StefanD986, 2015-09-03:
When calculating the impedance of a rectangular waveguide filled with a dielectric (Er != 1) the result is wrong. It looks like a factor of sqrt(Er) was forgotten in the equation: 

Example: given a rect. waveguide a=100mm, b=44.4mm filled with an Er=1.69 the wave impedance for the TE10 mode should be:
Z_F = (Z_0/sqrt(Er)) / sqrt(1 - lambda0 / lambdaCE)

with 
Z_0 = free space impedance (120Pi Ohm)
Er = 1.69
lambda0 = c_0/f
lambdaCE = lambdaC * sqrt(Er)
lambdaC = 2/sqrt( (m/a)^2 + (n/a)^2 ) = 2*a for the TE10 mode.

The expected result should be Z_F = 355 Ohm
Qucs / Qucstrans gives a value of Z_F = 461 Ohm for this example. I played around with the equation and it looks like that in the qucs source code Z_0 is not multiplied with sqrt(Er). 

Sources for the Z_F equation I used is "Taschenbuch der Hochfrequenztechnik, Band 2 - Meinke / Gundlach", Chapter K.4.3.
Side note to that: When the relative permeability is also taken into account I *assume* that the equation should be (but this case is not considered in my source):
Z_F = (Z_0*sqrt(Mur / Er)) / sqrt(1 - lambda0 / lambdaCE)    with lambdaCE = lambdaC * sqrt(Er / Mur)

For quick comparison this might be helpful: http://www.amanogawa.com/archive/RectWaveGuide/RectWaveGuide-2.html

I have tested this in qucs 0.0.18 qucstrans on OSX 10.10.5 (via homebrew). 
Example circuit:

```
<Qucs Schematic 0.0.18>
<Components>
<Pac P1 1 230 250 -74 -26 1 1 "1" 1 "50 Ohm" 1 "0 dBm" 0 "1 GHz" 0 "26.85" 0>
<Pac P2 1 410 250 18 -26 0 1 "2" 1 "50 Ohm" 1 "0 dBm" 0 "1 GHz" 0 "26.85" 0>
<GND * 1 230 280 0 0 0 0>
<GND * 1 410 280 0 0 0 0>
<RECTLINE RLTC1 1 320 200 -26 25 0 0 "100 mm" 1 "44 mm" 1 "101.6 mm" 1 "1.69" 0 "1" 0 "0" 0 "2.43902e-21" 0 "26.85" 0 "unspecified" 0>
<Eqn EqnTC1 1 380 360 -23 12 0 0 "A=twoport(S,'S','A')" 1 "ZL=real(sqrt(A[1,2]/A[2,1]))" 1 "yes" 0>
<.SP SPTC1 1 230 340 0 51 0 0 "const" 1 "0.2 GHz" 0 "20 GHz" 0 "[2 GHz]" 1 "no" 0 "1" 0 "2" 0 "no" 0 "no" 0>
</Components>
<Wires>
<230 200 290 200 "" 0 0 0 "">
<230 200 230 220 "" 0 0 0 "">
<350 200 410 200 "" 0 0 0 "">
<410 200 410 220 "" 0 0 0 "">
</Wires>
<Diagrams>
</Diagrams>
<Paintings>
</Paintings>
```

StefanD986, 2015-09-03:
Here the example file for Qucstrans:

```
# QucsTranscalc 0.0.18  /Users/stefan/Desktop/Untitled.trc
#   Generated on Do. Sep. 3 2015 at 17:28:17.
#   It is not suggested to edit the file, use QucsTranscalc instead.

<Rectangular>
  Er 1.69 NA
  Mur 1 NA
  Cond 4.1e+20 NA
  Tand 0 NA
  TanM 0 NA
  Freq 2 GHz
  a 10 cm
  b 4.4 cm
  L 4000 mil
  Z0 461.069 Ohm
  Ang_l 259.187 Deg
</Rectangular>
```

andresmmera, 2015-09-06:
Hello, 

I have not that book (nor I could find it on the Internet ), but from the Pozar's book (Page 113. Equation 3.86) I've found that Z_{TE_{1,0}} should be 354.66 Ohm, so it seems that you are right. 
I'll try to take a look to the waveguide component and see what's going on.

Note: I've employed the following [Sage](http://cloud.sagemath.com/) script to calculate the impedance:
freq = 2e9;#Frequency
lambd = 299792458/freq;#Wavelength

#TE mode
n=1
m=0;

# Rectangular waveguide dimmension
a= 0.1; #width (in meters)
b=0.0444#height (in meters)

#Permeability
mu0=4*pi*1e-7;#Vacuum
muR=1;#Relative
mu = mu0*muR;

#Permittivity
epsilon0 = 8.854187817e-12;#Vacuum
epsilonR=1.69;#Relative
epsilon = epsilon0*epsilonR;

k=2*pi*freq*sqrt(epsilon*mu);#Wave number
beta = sqrt(k*k - ((n*pi)/a)^2 - ((m*pi)/b)^2)#Imaginary propagation constant
ZTE = k*sqrt(mu/epsilon)/beta #Wave impedance. Pozar's book pg 113
print(N(ZTE))