Qucs/qucs #348: Unintsall/Remove QUCS on OSX

olr666, 2015-09-07:
I installed QUCS on my mac a while ago and have never really used it, so I am trying to uninstall it. I deleted all the qucs files I could find, but qucsedit is still in my launchpad and is undeletable.

Is anyone aware of a way to uninstall or remove qucs completely? I tried the uninstall shell but it doesn't work.

guitorri, 2015-09-14:
Which version did you install?

guitorri, 2016-01-23:
This should be fixed with #414.

Please open a Terminal app and try these commands one by one:

```
curl -ksO https://raw.githubusercontent.com/Qucs/qucs/master/qucs/contrib/pm/uninstall-qucs-pkg.sh
chmod +x ./uninstall-qucs-pkg.sh
./uninstall-qucs-pkg.sh
rm uninstall-qucs-pkg.sh
```

guitorri, 2016-01-23:
From feedback on qucs-help it seems to be fixed. Closing.

closed 2016-01-23 by guitorri

in3otd, 2016-07-17:
For reference, the OS X uninstall script has been moved, and it can now be found at
https://raw.githubusercontent.com/Qucs/qucs/master/contrib/mac/uninstall-qucs-pkg.sh
or equivalently at
https://sourceforge.net/p/qucs/git/ci/master/tree/contrib/mac/uninstall-qucs-pkg.sh?format=raw