Qucs/qucs #353: (linux) Subcircuits can't be added into a scheme if LANG=ru_RU.utf8

denizzzka, 2015-09-14:
workaround:
```
$ LANG=C ~/local/qucs/bin/qucs
```

ra3xdh, 2015-09-14:
I cannot confirm this bug. Could you please provide more detailed step sequence to reproduce it?
 I have Russian system locale `LANG=ru_RU.UTF-8` and perform the following sequence of steps:

1. Create empty schematic
2. Place empty subcircuit (File components->Subcircuit)
3. Go to subcircuit properties, then click Browse button and select appropriate subcircuit
4. Click Apply
5. Subcircuit is placed on schematic and simulation works.

Subcircuits in projects are also operational. So, it's need more details on your steps sequence.

denizzzka, 2015-09-14:
2) I am add try to add subcircuit by click on Content - Schematics - subcircuit_name and move mouse pointer to the sheet

On Russian locale it is do nothing, on LANG=C it is draws subcircuit symbol under mice pointer

ra3xdh, 2015-09-14:
Yes, I can reproduce this bug now. But there is one point that is missing from your comment. Qucs interface language should be set to "System default" in Application settings. If interface language is English all is operational. This looks strange. It's need to work on this bug.

ra3xdh, 2015-09-14:
I understood the reason of this bug. The problem is in `projectView.h` lines 31 -- 40. ProjectView tree labels lost their translations and could not be updated, because they are now wrapped in macros. But `tr("Schematic")` in line 2272 of `qucs.cpp` still have proper translation. It results in wrong `if` operator condition.  It's need to think how to fix it.

guitorri, 2015-09-14:
Can you try to cherry-pick this commit: https://github.com/guitorri/qucs/commit/da71cefe5f1d859cc0a07e3a6095b23043c31aa6

Those `tr()` as a define cannot be. This will break lupdate for the translations.

ra3xdh, 2015-09-15:
@guitorri , Yes, this commit helps, but it's need also to update all `*.ts` files. I have taken whole your `tx` branch and then updated again all `*.ts` files in it. There is also problem in `translations/Makefile.am` line 38. There is no backslash on end of this line.

 It seems all is already done on your `tx` branch. Can this `tx` branch be prepared for merge? What is need to do? As alternative we can backport individual commits from this branch in `master`.

guitorri, 2015-09-15:
I was working yesterday to update the tx branch. It is almost ready.
I will try to PR it later today.

guitorri, 2015-09-15:
@ra3xdh can you please take a look at https://github.com/Qucs/qucs/pull/357.

guitorri, 2015-09-17:
Seems to be fixed.

closed 2015-09-17 by guitorri

ldpgh, 2016-11-30:
I'm running into the same issue using Version 0.0.19 (9d924ca) @ Win7Pro, German
- Change of Application Setting  Language from "system language" to "English (en)"  & Restart seems to work
- Change of Application Setting  Language from "system language" to "Deutsch (de)"  & Restart does not work
- clone of Qucs/Github @ 2016/11/06-18:33:47 has the same issue
- Revisions 0.0.18 and before do not have this dependency

guitorri, 2016-11-30:
First, your issue seems different from the one reported here. Please open another ticked if that is the case.

Second, I cannot reproduce. I just tested switching languages with the `-rc3` and it seems fine.
Which branch are you using? The latest is `release-0.0.19`.

Please give it try with the portable version:
https://sourceforge.net/projects/qucs/files/qucs-binary/0.0.19-snapshots/qucs-0.0.19-rc3-win32-mingw482-asco-freehdl-adms.zip

ldpgh, 2016-11-30:
Have checked the portable version ... behaves the same as reported. It means language has to be changed to "English (en)" to get work.

guitorri, 2016-11-30:
Just to be clear, you cannot add sub-circuits if you set the language to German. Right?
Your Windows is in German, right?

I can add sub-circuits with German set, but my windows is in English... Still trying to reproduce.

ldpgh, 2016-11-30:
You got the right impression. 

In the meanwhile have tested the mobile Version on my old laptop running WinXP + German. It works for all 3 mentioned conditions.

One remarkable point is application language setting in case of German.
- Win7Pro ... "Deutsch (de)"
- WinXP ... "German (de)"
It's not what I would expect.

ldpgh, 2016-11-30:
Mobile Version ...
- https://github.com/Qucs/qucs/issues/353#issuecomment-140061759 ... OK ... same as reported
- https://github.com/Qucs/qucs/issues/353#issuecomment-140121777 ... issue ... same for me as reported above

in3otd, 2016-11-30:
uhm, quite strange. Also here, with Windows with English system language and Qucs language set to German it works fine. I'll need to try to switch the system language to German, not sure if it's possible.
When you wrote
> 
    Win7Pro ... "Deutsch (de)" ... issue
    WinXP ... "German (de)" ... OK

you mean that in both cases the Qucs messages are shown in German, right? Just to be sure the application is selecting the right language in both cases.

guitorri, 2016-11-30:
Just to say, there is no `Deutch` string in the repository... so I don't understand.

ldpgh, 2016-11-30:
Screenshot Application Setting @ Win7Pro
[image]

guitorri, 2016-11-30:
Something is wrong:
[image]

I set it to German and restarted. Parts of the UI are in German parts are in English (as my Windows)

guitorri, 2016-11-30:
I thing there is a bit mess up in the translation files, they are not translated anymore... :angry: 
https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/translations/qucs_de.ts#L12647-L12648

ldpgh, 2016-11-30:
@guitorri ... New ticket or do you reopen this one?

in3otd, 2016-11-30:
uhm, it seems that many German translation strings were already lost here...
https://github.com/Qucs/qucs/commit/36abe569f763740be8a2c484389be4544d280b7c#diff-3863895ef601ae42db826aa712bbb23c

@ldpgh , do you have an "old" version of Qucs installed on one of the Windows machines but not on the other? I think the new version may be using the translated strings of an older installed version (may happen if there is a `QUCSDIR` environment variable set, I think)

in3otd, 2016-11-30:
...my wild guess:
- on the Win7Pro machine you have an old version of Qucs installed, with the environment variable `QUCSDIR` set, so Qucs points to the translated strings files of the old version. The old translations files do not contain all the strings used in the new versions (of course) and in particular not the translated string used in the comparison [here](https://github.com/Qucs/qucs/blob/release-0.0.19/qucs/qucs/qucs.cpp#L2324) so the check fails.
- on the WinXP machine no older versions are installed (or only a very old one), so that `QUCSDIR` is not set and so the portable Qucs version used the updated translation files and you see the correct behavior as we see here

ldpgh, 2016-11-30:
Hello Sherlock Holmes, your guess is exceptional good description of my setup.

All your assumptions are true. Though the WinXP machine have seen old Qucs installations, all were de-installed recently. And yes QUCSDIR is set on the Win7Pro machine.

Thx for your quick support. No further work should go in this topic and my focus returns to PR for #386.

guitorri, 2016-11-30:
So, (re)setting QUCSDIR fixes the sub-circuit issue?

We still have two issues. The lost translations and comparison done on translated strings...

guitorri, 2016-11-30:
I did a bisect on the German word `Englisch`.
Something like ` find . -name qucs_de.ts | xargs grep Englisch`
A bunch of German translations were lost before 0.0.17, in this commit 6663635cf

@ldpgh please confirm that the sub-circuit issue is fixed if QUCSDIR is correctly set.

in3otd, 2016-11-30:
> We still have two issues. The lost translations and comparison done on translated strings...

yep... I'll add a third one: the portable Qucs uses `QUCSDIR` unconditionally.
It is/will be a common situation that a Windows user, which has previously installed Qucs, uses the portable version without uninstalling the old version first. The result will be that some/many things will be taken from the old version, as seen above. IMHO, the portable version should have a `qucs.bat` file in the base directory that unsets `QUCSDIR` and then launches Qucs. And/or even better a startup message that inform the user that `QUCSDIR` is set and this should not normally be.

I didn't look at the installer but I think it should make sure that `QUCSDIR` is either not set (best) or set to the newest Qucs (but this may break the old install).

ldpgh, 2016-12-01:
QUCSDIR (both settings (user & system)) removed. Now portable-Qucs operates properly.

ldpgh, 2016-12-01:
Adding on the long-term a startup-wrapper script (e.g. qucs.bat) is a good idea ... and not just for windows. IMHO it's the right way for CAD software running on many platforms and systems with old versions installed.

guitorri, 2016-12-01:
Thank you @ldpgh for the update. There are now tickets open to track the individual issues.