Qucs/qucs #362: Unused variables in cpwline, can remove?

guitorri, 2015-09-17:
I don't know if these unused variables are indeed not necessary. Can someone check?

```
C:/Users/me/git/qucs/qucs-core/src/components/microstrip/cpwline.cpp: In member function 'void cpwline::initPropagation()':
C:/Users/me/git/qucs/qucs-core/src/components/microstrip/cpwline.cpp:165:20: warning: variable 'se' set but not used [-Wunused-but-set-variable]
     nr_double_t d, se, We, ke, qe;
                    ^
C:/Users/me/git/qucs/qucs-core/src/components/microstrip/cpwline.cpp:165:24: warning: variable 'We' set but not used [-Wunused-but-set-variable]
     nr_double_t d, se, We, ke, qe;
                        ^
C:/Users/me/git/qucs/qucs-core/src/components/microstrip/cpwline.cpp: In static member function 'static void cpwline::analyseQuasiStatic(double, double, double, double, double, int, double&, double&)':
C:/Users/me/git/qucs/qucs-core/src/components/microstrip/cpwline.cpp:311:20: warning: variable 'se' set but not used [-Wunused-but-set-variable]
     nr_double_t d, se, We, ke, qe;
                    ^
C:/Users/me/git/qucs/qucs-core/src/components/microstrip/cpwline.cpp:311:24: warning: variable 'We' set but not used [-Wunused-but-set-variable]
     nr_double_t d, se, We, ke, qe;
                        ^
```

andresmmera, 2015-09-17:
I've taken a look and they can be removed for sure. It seems that these variables were used for testing an expression, but this expression is no longer used... Here is the piece of code that is causing trouble:

nr_double_t d, **se, We** ke, qe;
    d  = (t * 1.25 / pi) * (1 + qucs::log (4 * pi * W / t));
    **se = s - d;**
    **We = W + d;**

    // modifies k1 accordingly (k1 = ke)
    //ke = We / (We + se + se); <='We' and 'se' only appear here, so they can be removed**
    ke = k1 + (1 - k1 * k1) * d / 2 / s;

Just in case, I've simulated the s-parameters of a CPW before and after suppressing 'We' and 'se' and the results are identical. I've created a branch with the correction https://github.com/andresmmera/qucs/tree/CPW_Unused_Variables

guitorri, 2015-09-17:
Thanks! I created the PR #365 for you ;)

closed 2015-09-17 by guitorri

in3otd, 2015-09-17:
(just saw this now, for some reason I do not receive notifications for many of the updates ??)

FYI, I'm probably at the origin of the unused variables warnings, with PR #306 . The commented out expression for `ke` was supposed to give identical results as the other expression (for small `d`) but did not and I didn't have the time to investigate more (both formulas are seen in texts dealing with CPW impedance calculations)