Qucs/qucs #367: LM317 models not working as expected

ntavish, 2015-09-18:
[image]
Vout should be 2.5V, and it should settle above around Vin=5V

Parameter sweep on R2, with constant Vin also is not at all as expected.

guitorri, 2015-09-18:
Can you put the schematic into a Gist, or somewhere we can have a look a it?

ntavish, 2015-09-18:
https://gist.github.com/ntavish/952c281f78be632c9f7a

This isn't on the same computer, and probably an older version(0.0.15-1.1 version for ubuntu 12.04), but result is the same.

ra3xdh, 2015-09-18:
I think it is bug on Qucs-library side. There are two models for LM317 in Qucs library. LM317K-2 is working as expected (2.5 V at output), LM317K-1 and LM337 give wrong result. @ntavish , You can use LM317K-2 to get proper result.
[image]

in3otd, 2015-09-18:
looking at the component netlist I have the impression that the ADJ and OUT connection were swapped; indeed when swapping them on the schematic the output voltage seems correct.

in3otd, 2015-09-20:
BTW @ra3xdh , the LM3**3**7 is a *negative* voltage regulator, if you apply -12 V at its input it works as expected.

ra3xdh, 2015-09-29:
Closed via #375.

PR #375 (2015-09-29): Correct pin assignment in LM317 regulator
Fixes #367. The pins OUT and ADJ were swapped in the LM317 model. Note that the model has/causes some convergence issues. 
The two LM317 model in the library are very similar but not identical (a few BJT have different area and/or saturation current), so some results, like the exact output voltage or the maximum current are different.

Simulations results below, with the two LM317 models (and with the LM337 just to show it works well, with a negative voltage at input)

[image]

[image]
closed 2015-09-29 by ra3xdh