Qucs/qucs #372: Hardcoded .qucs paths

in3otd, 2015-09-19:
The "old style" resource files and path are still used by Qucs Help [here] (https://github.com/Qucs/qucs/blob/master/qucs/qucs-help/main.cpp#L51), so that it always saves an `helprc` file in a `.qucs` directory, even if the QucsHome has been defined to be elsewhere, and reads back the settings from the same place. As a side effect, Qucs Help does not use the language selected in the main Qucs GUI.

Similarly Qucs Transcalc still tries to load a startup file from a fixed location [here] (https://github.com/Qucs/qucs/blob/master/qucs/qucs-transcalc/main.cpp#L135) and sets the initial working directory to a [fixed path] (https://github.com/Qucs/qucs/blob/master/qucs/qucs-transcalc/main.cpp#L116) independently of QucsHome.

felix-salfelder, 2015-09-28:
is this related to #335?

in3otd, 2015-09-28:
no, not really, in the sense that this likely won't fix #335, also because the root cause is not clear.
In the error messages reported there, there is a `Can't open file NONE/share/qucs/docs/en/index.html` and I think the only way to have that is to compile with a DOCDIR set to `NONE`, see [here] (https://github.com/in3otd/qucs/blob/issue_372/qucs/qucs-help/main.cpp#L135), and I do not see how this could happen.

guitorri, 2017-02-17:
I think this can be closed.
Qucs-help is being removed.
Transcalc was fixed in 1fe13bb635cd

closed 2017-02-17 by guitorri