Qucs/qucs #374: wrong run time paths

felix-salfelder, 2015-09-20:
when running qucs out of `$(builddir), it cannot execute the other tools, this is because of some QucsBinDir hack, that does not seem to make sense in all cases.

looking from further up it seems that the run time path philosophy currently is

- qucs needs to be installed to work properly
- external tools (such as admsXml, octave) don't need to be installed, but some environment variables point to them (their build location maybe?)

instead i propose to pursue the following:

- qucs does not have to be installed in order to work properly.
- external (3rd party) tools must be installed, so qucs works right with default settings (other settings may still be possible, yet unneeded).

a package is "installed" when it can be invoked (and works) from the usual (unmodified) user environment). this typically is the case after "make install" has been run, or a package manager has installed a package (apt-get, yum and the like).

ra3xdh, 2015-09-20:
* `$QUCSDIR` setting helps for non-standard cases. I doubt that your proposal is necessary.

* 
> external (3rd party) tools must be installed

What do you mean? We will have bloated dependencies of Qucs Linux package in such case.  3rd-party software should be optional.

in3otd, 2015-09-20:
uhm, I do not see what's wrong in having to install a set of tools to have them properly working together. AFAIK, you can run every single tool without installing, but if you would like the Qucs GUI to launch the other tools, it needs to know where they are. The Qucs GUI assumes all the Qucs tools are in the same `bin/` directory, why shouldn't be so? Then it needs to know where are the others, external, tools so they need either to be in the `$PATH` or to have their location specified otherwise.

What is the actual problem that needs to be solved here ??

felix-salfelder, 2015-09-20:
> uhm, I do not see what's wrong in having to install a set of tools to have them properly working together. 

exactly. this is why ADMSBINDIR, OCTAVEBINDIR and the like are redundant. properly installed tools dont need this.

> The Qucs GUI assumes all the Qucs tools are in the same bin/ directory

before `make install` they simply are not, so the assumption is wrong. `make check` must work before `make install`.

> What is the actual problem that needs to be solved here ??

try `Ctrl+4` in qucs gui, ran from its builddir. it will fail (if you did not `make install`). worse, it will run a previously installed tool, potentially unrelated to the current build tree,

> We will have bloated dependencies of Qucs Linux package in such case. 3rd-party software should be optional

@ra3xdh sorry i was imprecise here. "must be installed" only refers to the case that they are meant to be used. the emphasis is on *installed*, rather than "not installed" = "located at random places to be figured out by the user himself". hope you can agree now.

in3otd, 2015-09-20:
> exactly. this is why ADMSBINDIR, OCTAVEBINDIR and the like are redundant. properly installed tools dont need this.

to use the ones in the current path, if found, could be the default behavior. Still allowing to be overriden by environment variables or applications setting... Qucs is flexible... too flexible.

> before make install they simply are not, so the assumption is wrong. make check must work before make install.

as said, the Qucs suite is not meant to be used from the build directory. Seems reasonable to me to have to install a (set of) SW before using it.
`make check` *will* work before `make install`, if not asked to do more than what is supposed to do. It should check the individual programs, standalone, for what is possible, not the interactions between the entire suite. For doing more you should likely use the `distcheck` and `installcheck` targets.

felix-salfelder, 2015-09-20:
> as said, the Qucs suite is not meant to be used from the build directory.

@in3otd would it bother you if i did anyway? to me it's just not properly implemented.

> make check *will* work before make install

currently, yes. because the test suite is incomplete.

> For doing more you should likely use the distcheck

distcheck runs `make check`, but you knew this, i don't see your point.

> and `installcheck` targets.

`make installcheck` implies `make install`. so it does not avoid the install target. `make install` is destructive, i will certainly not want to run it on random pull requests just to see if something is (not) working some way or other. apparently some do, but i can only fix the code.

in3otd, 2015-09-22:
> @in3otd would it bother you if i did anyway? to me it's just not properly implemented.

it's just the usual difference of opinions. To me the fact that the Qucs tools do not work directly from the builddir is a non-issue and that you can make them work from there is a weak reason to have it merged.

See, at present Qucs [cannot read mail] (https://en.wikipedia.org/wiki/Jamie_Zawinski#Zawinski.27s_law_of_software_envelopment) but surely could be expanded to do so. Shall we accept a working PR for this? We should draw the line somewhere.

Users will install Qucs to use it and the directory structure of an installed Qucs is not the one used by the source. To test what the users will "see", Qucs has to be installed.
I might even say that it's actually good that the Qucs suite cannot be fully run from the builddir, in this way it will be tested only in the final configuration a user will see, after `make install`.

Regarding the `make check` et al., I'm sure you are able to convince an autotools target to do something that another target is designed to do but, as I see it:
- `make check` runs test on the individual programs. E.g. it tests that the Qucs GUI can convert a schematic to a netlist, that `qucsconv` can convert between a few formats, simple things like these, which do not involve interactions between the various programs
- `make distcheck` tests... the whole distribution, if the programs interact as expected. And, yes, it implies a `make install` but, of course, the install is automatically *done in a temporary random directory*. It will automatically run `make installcheck` which will do test involving more than one Qucs program. A good way to check if Qucs can find all of its tools in any case, no? And I see nothing destructive in this.

BTW, all the PR I check involve a `make install`, in this way I'm in the situation the end user will see, why should this be avoided? Of course the install is in local directory, outside of the `$PATH`. I have a few Qucs (Qucses?) installed locally at the same time, plus the main, system-wide, install, this never caused a problem.

For our affectionate readers, here is a link about what [`distcheck`] (http://www.gnu.org/software/automake/manual/html_node/Checking-the-Distribution.html) is meant to do.

And let me quote [a paragraph of the Autotools manual] (http://www.gnu.org/software/automake/manual/html_node/Basic-Installation.html):
> A last and optional step is to run `make installcheck`. This command may run tests on the installed files. `make check` tests the files in the source tree, while `make installcheck` tests their installed copies. The tests run by the latter can be different from those run by the former. For instance, there are tests that cannot be run in the source tree. Conversely, some packages are set up so that `make installcheck` will run the very same tests as make check, only on different files (non-installed vs. installed). It can make a difference, for instance when the source tree’s layout is different from that of the installation. Furthermore it may help to diagnose an incomplete installation. 

IMHO this applies perfectly to the Qucs situation.

felix-salfelder, 2015-09-22:
> See, at present Qucs cannot read mail but surely could be expanded to do so. Shall we accept a working PR for this? We should draw the line somewhere.

here, you are making fun of something. but i don't quite get it. my PR is about *fixing* something, not implementing some extra feature (such as read mail).

the issue would not even arise, if things were properly implemented in the first place -- why do you make up a conflict of interest on this?

> Users will install Qucs to use it and the directory structure of an installed Qucs is not the one used by the source. To test what the users will "see", Qucs has to be installed.

i am not talking about users. this is about development and testing. if you carefully read my patch, you will notice, that it does not change the situation after `make install`. go figure, this might be intentional.

> And let me quote a paragraph of the Autotools manual:

talking about autotools ... please explain, why qucs-core/src/qucsator is a wrapper script.

ra3xdh, 2015-09-22:
@felix-salfelder ,  I also support @in3otd 's point of view. I also think that Qucs should be tested in final configuration as user will see it. Current Qucs installation system is  flexible enough. We don't need to support all non-standard cases of Qucs run. Current build and deploy system can satisfy all testing cases using `$QUCSDIR` environment variable. It's not so complex to set this environment variable and installation type in your favourite IDE.

>  my PR is about fixing something
Something should be and may be fixed only if it is broken. At current state there is nothing broken in Qucs build system. So, your PR is redundant enhancement. 

Concerning #376 , I think this PR should not be merged, because all user-space installation cases could be processed via `$QUCSDIR`. This PR brings more problems than solutions. For example you have introduced system variable `$QUCSTRANS`. What was need to introduce it just for Qucs-transcalc? Why all other utilities don't need such variable?

felix-salfelder, 2015-09-22:
> I also think that Qucs should be tested in final configuration as user will see it.

feel free to test as much as you will. my PR does not get into your way. (but you knew this).

> I think this PR should not be merged.

please think again. take into account that your practises and customs do not fit everybodys needs.

> For example you have introduced system variable $QUCSTRANS

read the patch again. QUCSTRANS is an environment variable, it has nothing to do with your system. if you are against proper environment variable semantics, we could set PATH instead, and not even need `QUCSTRANS`. but this is a different topic, just open a new thread.

@ra3xdh we had a similar discussion about the QUCSATOR environment. the positive outcome was hard and unnecessary work on my side. i have a felling the you are trying to get rid of me in particular. open an issue, get me out of the project, but don't interfere with my work like this. i have other things to do.

in3otd, 2015-09-22:
> here, you are making fun of something. but i don't quite get it. my PR is about fixing something, not implementing some extra feature (such as read mail).

No, I'm not trying of making fun of anything or anyone. I just took a random example that I hoped was clear for anyone that is should not be in Qucs.
If you wish, you can see this PR as adding an extra feature to the development process; this feature is simply not needed and can even be misleading, allowing to use the Qucs suite in a way that no user will ever do. And I see no advantage for the developers either.

If the problem you are trying to solve is "cannot fully test the Qucs suite from the build directory" the solution is simple. Test it the proper way and install it before testing.
Your PR doesn't get in the way of the usual testing, sure, but the fact that it works to solve your particular problem and can be ignored by the others does not make a good reason to merge it.

Back to something more constructive, as you said ADMSBINDIR and OCTAVEBINDIR should not be needed for the "usual" cases, they should be there just to allow experimenting around for the more advanced users. At present we are using hardcoded paths, we should likely check at the firs run if these are available in the `$PATH` and if not inform the user to change the Qucs settings and/or set the external variables.

`make distcheck` currently fails due to many little things missing from the Makefiles; I'm taking a look at this.

felix-salfelder, 2015-09-23:
> No, I'm not trying of making fun of anything or anyone.

good. but it was insanely funny.

> I just took a random example that I hoped was clear for anyone that is should not be in Qucs.

yes, i am totally convinced that your example should not be in qucs (no irony).

> Back to something more constructive, as you said ADMSBINDIR and OCTAVEBINDIR should not be needed for the "usual" cases.

i couldn't care less about ADMSBINDIR, OCTAVEBINDIR and the like. but they make good random examples for the schizophrenia in your argumentation, so lets try to fix your position one last time.

from admsXml, you *expect* that it works when *you* run it from its `$(builddir)`. this makes totally sense, as otherwise, *you* could not just build it (either embedded, or in some place pointed to by this ominous variable) and instantly use it to translate your code. then, when it comes to `qucs`. *you* (the same you) claim that just invoking `qucs` is the last thing somebody would wish to do. what a _u_niversal point of view (yes, very funny).

it may be the last thing *YOU* want to do. i see that, i acknowledge that, and i don't care. (i don't want to change your principles). but you *imply* that something that *YOU* don't need *MUST NOT* work. what if others (the admsXml authors) were as stubborn as this? would you not argue to get their code fixed?

i already pointed out that `qucs-core/src/qucsator` is a wrapper, do you understand why? automake/libtool is *designed* to just work for everyone, in any situation. this is not limited to cutting down the requirements to some arbitrary make and a posix shell. also, the run time semantics are chosen very sensitively. there is a rationale behind it. one might be the principle of least surprise. what (as a user or a developer) is the least surprising to happen when you start a program? will it work? will it not work? or will it work half way and suddenly break upon pressing `ctrl+4`, dare you! (guess what the autotools developers do upon a bug report without patch. indeed, i was surprised -- and i didn't even had to insist that it was a bug.)

next example. what if i don't have write access in `/usr/local`? heck that sounds made up. but it's not. be it a student that works at a computer he does not own. (no (s)he won't quickly install "ubuntu" or whatever on his/her phone.) you expect that (s)he knows all about how `--prefix` works and how he will certainly master his environment settings. look at the code: some of the `qucs` developers do or did not know about what `$PATH` is good for, even today...

> And I see no advantage for the developers either.

rethink. as I already wrote, `make install` is destructive. spelt out, this means that there is no inverse operation. read: there is no undo button. not depending on operations without an undo button is clearly a benefit. about safety: not installing something is safer. put random stuff into your `$PATH`, i'd better not. determinism. start `qucs` today, start qucs tomorrow (on another machine?). what was the environment again? does not matter, if you run from builddir. which advantage are you pretending not to see?

> Test it the proper way and install it before testing.

i will likely not do this, and i promise, you are not the one to change that. if `qucs` is broken by design, as you suggest, i'd rather resign in an attempt to fix the design. then in an attempt to fix the co-developers. software does not develop by itself or by adoration of its current state.

> oops, again, I forgot you have already a PR on this...

makes a good example, as well. *I* do the work, or some of it. let me choose *how* i work. it's none of *your* private business. read about my goals (here: fix a bug, run qucs). if you can't provide alternative ways, then just accept mine. don't tell me that it's not a bug once again -- be sure i wouldn't have bothered to report it.

guitorri, 2015-09-23:
+1 for consistency. 
If we allow environment variable override to some we should do the same for **all** tools external to Qucs. 

If we provide env. variables we should also provide ways to change them in the Settings dialog.
Perhaps something [like this](http://i.stack.imgur.com/YWIpR.png) from another context.

I am not sure anymore about the reasoning behind QUCSDIR. I might have something to do with not having to modify the PATH. On Windows, often you need admin rights to change the PATH. I guess you could use a user variable to work around that. Not sure anymore with new version of the OS. In any case, for some reason the installer still requires admin rights (which I don't think are really necessary). 

In the particular case of testing `ctrl+4`, often I am working only with qucs (with CMake and QtCreator) and use QUCSDIR to point to an existing installation if I need any of the external tools. 

Frankly, I don't see a problem with enabling the run form `$(builddir)` use case.

I like the fact that I can have multiple install prefixes without messing with the PATH. I guess `QucsBinDir` is sort of a portable way of doing it. 

We should document these things. We should not have to look into the code to know what overrides what...

@felix-salfelder why do you come back to the wrapper thing? Are you referring to the libtool wrapping mechanism? I don't get your message here. Are you referring that you are influencing only the environment for that wrapper?

On the premise that `$(builddir)` should work, I believe the translations need to be fixed as well. If I recall correctly with autotools the .qm files are only generated along with `make install`.

felix-salfelder, 2015-09-23:
@guitorri i appreciate your interference.

> If we provide env. variables we should also provide ways to change them in the Settings dialog.
Perhaps something like this from another context.

i cannot think of a use case. environment variables are something, the environment should provide (hence the name). the (texmaker) example is about inhomogeneous invocation of (external) tools, with a bias on configuration. for qucs, we simply need locations of (internal) executables. and only in cases where `$PATH` fails. here, we hardly (the goal is: *never*) need manual interaction.

> I am not sure anymore about the reasoning behind QUCSDIR. I might have something to do with not having to modify the PATH.

i think, the reasoning is, that (maybe because qucs does not work from where it was built?) people started to produce different installations with various prefixes unrelated to `PATH`. the QUCSDIR "hack" makes sure, that a consistent set (=all belong to the same installation) of tools is used. we obviously must stick to this "behaviour", or at least simulate it somehow...

my current approach, the easiest of all, just sets `QUCSATOR` (and others), in case qucs is ran from `$builddir`. qucsator is then invoked as `(isset($QUCSATOR))?$QUCSATOR:QucsBinDir/qucsator`. this solves most problems at once (remember the alternative qucsator...), is evidently backwards compatible, and totally foolproof.

> In the particular case of testing ctrl+4, often I am working only with qucs (with CMake and QtCreator) and use QUCSDIR to point to an existing installation if I need any of the external tools.

so you *do* run qucs out of builddir? funny. then, my proposed wrapper should not touch the environment in case `QUCSDIR` is set. i will add this for instant backwards compatibility.

> I like the fact that I can have multiple install prefixes without messing with the PATH. I guess QucsBinDir is sort of a portable way of doing it.

i'm (still, currently, for now) against removing that "feature". however, there might be simpler ways to achieve the same. the most robust that comes to mind is this. compare the outcome of `which qucs` against `which argv[0]`. if it is unequal, prepend `dirname($0):` to `$PATH` (some time in early `main()`) and be done. if there's no portable solution to modify env, use a wrapper (imo: ugly. look what others do, iceweasel maybe).

right now, i'd like to not touch it, and work around, where necessary. the better way would certainly be, to not use executables at all. what is happening here is a case for `dlopen`. but the transition to `dlopen` is going to be some more work (not counting in the discussion).

> why do you come back to the wrapper thing?

i was runnig out of arguments. the wrapper is an example where people have gone a long way to make things work right (for everyone!). it shows how silly it is to define a new standard for a particular project just out of personal habit.

my implementation is inspired by the libtool wrapper, but completely independent. there does not seem to be a good way to reuse/tweak the existing libtool wrapper.

> On the premise that $(builddir) should work, I believe the translations need to be fixed as well. If I recall correctly with autotools the .qm files are only generated along with make install.

this is trivial, i just didn't notice, thanks for pointing out.

in3otd, 2015-09-24:
uhm , still I do not get why setting `QUCSDIR` and testing the Qucs *suite* in a files tree which is not is the install one is a Good Thing. In this way one explicitly avoids using the current code for everything but the GUI. 
If one day we will have 10x the currently active developers it will be fine that everyone focuses only on its area of interest and maybe 2 or 3 take care of the integration but with just 3 or 4 active people it does not make sense to look  at a small part only; my feeling is that there are already too many things which are not really tested by anyone anymore.
IMHO, Qucs is already too flexible and I do not find unreasonable to ask the users/developers to adapt to Qucs instead of the other way around in some cases. 

Yet the Project Leader said we can support running from the builddir, so, ok, we will have that.

Still, why one should avoid to use `./configure --enable-maintainer-mode --prefix=/home/$(whoami)/local/qucs/` and `make install`as we even recommend on the wiki, still remains a mystery to me. And yet, doing this way, there is no need to change the `$PATH`, no need to fiddle with `$QUCSDIR`, it just works, as it should. Why should `make install` in a user local directory destructive? 

I know that the student with no write access to `/usr/local` is a common situation, *I* usually do not have write access to `/usr/local` on my own system here and surely I do not need it to use Qucs.
The fact that ey might not know how to use `--prefix` seems a little made up to me. If ey does not know yet, is not willing to google it, does not try to look at the top-level `README`, it won't probably even find out that ey has to type `qucs/qucs` to launch it.
Yet, support also the poor student: let `configure` print out a message at the end, telling explicitly how to change the install directory w.r.t the default defined one. Add a compilation hook to so that at the end of the compilation a short message telling what ey can do is printed: "run `qucs/qucs` to use Qucs without installing, run `make install` to install into the $prefix directory" (already seen somewhere).
And, yes, ey should not have to try running the tools to discover they cannot be found, there should be a check at startup if every tool can be run and informed otherwise, starting from builddir or not.

On the origin of `QUCSDIR`, I do not know for sure, but I don't think it had originally anything to do with development, I have the impression it was originally introduced to support relocatable binary packages (anyone needs/uses that here?)

felix-salfelder, 2015-09-24:
> uhm , still I do not get why setting QUCSDIR and testing the Qucs suite in a files tree which is not is the install one is a Good Thing.

i'd highly appreciate a PR that removes `QUCSDIR`. relying on PATH only will simplify this PR quite a bit. i expect serious (and justified) contradiction. lets do it the hard way (transitions always are hard): implement this PR, then change habits (making QUCSDIR obsolete). and *then* remove QUCSDIR. if the latter does not succeed, bad luck.

> Yet the Project Leader said

come on, this is not a dictatorship.

> Still, why one should avoid to use `./configure --enable-maintainer-mode --prefix=/home/$(whoami)`

1st: there's no guarantee that you have write permission in `/home/$(whoami)` (try $HOME instead). second: there's no reason to avoid that. it's just pointless (unless running in place is broken by some design).

telling people to `install` in random places is missing the whole point of the `install` target. spreading practices typically leads to mechanical adoption of these practices. go ask a random student why he typed/pasted `make install` at all. he might answer: "i have been told to do so" (or something similar), rather than: "the files of my favourite programs are sorted by role into a tree rooted somewhere in my home directory, so i can (whatever) ... ".

> Why should make install in a user local directory destructive?

it *destroys* your (previous) installation, no matter what the prefix is.

> "run qucs/qucs to use Qucs without installing, run make install to install into the $prefix directory" 

which is supposed to be `qucs/qucs/qucs`. uhh, how about turning that into `qucs-gui/src/qucs`? (good point anyway.)

felix-salfelder, 2015-09-25:
> On the premise that $(builddir) should work, I believe the translations need to be fixed as well. If I recall correctly with autotools the .qm files are only generated along with make install.

@guitorri rechecked. cannot reproduce. here, `make -C translations clean` deletes all .qm files, and `make -C translations` regenerates them.

however i will have to make sure that qucs is able to locate these files at runtime. shouldn't i get an error in case they are not in place? where/when are the files actually used?

in3otd, 2015-09-26:
> I'd highly appreciate a PR that removes QUCSDIR. relying on PATH only will simplify this PR quite a bit

oh! Here we seem to agree :grin: ! AFAIU, `QUCSDIR` could be immediately removed for the Linux environment. Not sure about Win and Mac. But this might mean a change of habits for some developer... 

> >    Yet the Project Leader said

> come on, this is not a dictatorship.

sure, it's not, but there should be some "mechanism" to end a dispute (I said something similar when we discussed of the review process, I think). Having someone setting the direction at least makes clear where the project is going, better than having people with different opinions continuously pulling it in different directions. And since guitorri surely knows the project better than me, makes sense he sets the direction, to me.

felix-salfelder, 2015-09-26:
> oh! Here we seem to agree

almost. imo, if somebody makes use of QUCSDIR, (s)he should not be stopped. QUCSDIR implements some sort of relocation. my approach provides a (much impler) relocation (just don't `make install`), as a substitute.

guitorri, 2015-10-02:
QUCSDIR might still be useful (not sure anymore), but should not be set by the Windows installer (need to fix that). As of now (with QUCSDIR set) so we cannot have two installed versions side by side on Windows.

felix-salfelder, 2015-10-02:
if, i mean *if*, we want to remove QUCSDIR, then the 0.0.19 release should emit a deprecation warning if it is set. imo not within the scope of this PR.

felix-salfelder, 2020-08-31:
this seems all fixed with #376

closed 2020-08-31 by felix-salfelder