Qucs/qucs #382: question regarding symbols

felix-salfelder, 2015-09-26:
there is an "edit properties" context menu entry for symbols (such as instances of lumped\ componente->resistor).

the edit properties menu has "add" and "remove" buttons, however, they are inactive (greyed out). what is the rationale behind this?

what i would like to do is, instanciate a symbol, and represent a component of my choice. for example, add a "COMPONENT" attribute to the resistor symbol, with value "my_fancy_nonlinear_resistor".

in a very general sense i think it makes total sense to permit (user defined) parameters attached to symbols. i am wondering why this is not implemented already.

sure, in addition, i'd like to point to an implementation of my nonlinear resistor, maybe using a "FILE" or "LIBRARY" attribute. or maybe make use of a component library attached to my schematic. (note that this is not what i am asking here.)

in3otd, 2015-09-27:
the "add" and "remove" buttons are enabled for the `Equation` component (and maybe for another few special cases?), so that you can add/remove new equations there. As you saw, most of the other components do not support adding new properties.

ra3xdh, 2015-09-27:
> As you saw, most of the other components do not support adding new properties.

It's not quite in a such way. User components are represented in Qucs as subcircuits. New properties could be added/removed only for user components only at subcircuit creating stage. `File->Edit Circuit Symbol`, then double click on component name. Here one can add/remove new properties for user defined components. Then you can build user library form subcircuits.

> what i would like to do is, instanciate a symbol, and represent a component of my choice

> or maybe make use of a component library attached to my schematic.

@felix-salfelder , All this stuff is already implemented in Qucs. But there may be empty gaps in documentation. You should use combination of subcircuits and user libraies. You need to perform the following steps:

* Create a project containing a set of subcircuits. Specify desired parameter and symbols for every  subcircuit. Subcircuits could be directly used in main schematic, or used for making user libraries. 
* Then use `Project->Create Library` to create user library. It seems this process is not documented (?)
* This library will appear in QucsLib tool and could used with system libraies.
* To embed user libraries in Qucs project package `.qucs` use `include user libraries` checkbox in `Create Package` dialog. 

Two thing that may need to implemented in this process:

* Component symbol patterns. It's need to allow user to use predefined symbol pattern (R,C,BJT, etc.) instead of drawing symbol by hands. This was suggested by @in3otd  It will allow to make user components development more rapidly.
* I have not yet tried user library consisting of Verilog-A defined components. Is it possible?

felix-salfelder, 2015-09-27:
i think @in3otd explained everything, thanks.

> All this stuff is already implemented in Qucs.

@ra3xdh i cannot confirm. i only see greyed-out buttons. so lets make use of them!

> I have not yet tried user library consisting of Verilog-A defined components. Is it possible?

haven't looked into libraries yet. ideally, a library should be/contain a list of references to files/directories. then, Verilog-A suddently is no longer a special case.

ra3xdh, 2015-09-27:
User components parametrization is described in `Help->Tutorials->equations.pdf` It's strange that you have not met with this process before. Did you ever use user-defined components?

> Verilog-A suddently is no longer a special case.

It seems it is a special case. Such library may require embedding of compiled modules. I have just tested it. Library is successfully created but schematic don't operate properly, because compiled modules are not attached to library components.

felix-salfelder, 2015-09-27:
> Did you ever use user-defined components?

you are kidding. but no, i havent ever encountered anything as entangled as qucs.

> Such library may require embedding of compiled modules.

certainly not.

in3otd, 2015-09-27:
> > As you saw, most of the other components do not support adding new properties.

> It's not quite in a such way. User components are represented in Qucs as subcircuits.

@ra3xdh you are of course right, I was thinking at which other components enable the buttons "add" and "remove" buttons in the "Edit Component Properties", which is used only for "standard" components, AFAIK.

closed 2015-09-27 by felix-salfelder