Qucs/qucs #384: Suggest noting dependency on FreeHDL for digital simulations

danyill, 2015-09-27:
I suggest adding a note to installation/build instructions that for for digital simulations FreeHDL must be installed and that this can occur in Debian/Ubuntu with the following command

`sudo apt-get install freehdl`

If FreeHDL is not installed, then the following error message results when carrying out a digital simulation: `/usr/local/bin/qucsdigi: 1: /usr/local/bin/qucsdigi: freehdl-config: not found` however the pre-requisite for a correct installation should not be responding to an error message.

danyill, 2015-09-27:
Also, for digital simulations to get an executable "libtool" the following package needs to be installed in Debian Jessie at least:

`sudo apt-get install libtool-bin`

Perhaps this also could be added to installation instructions/dependencies?

arpit-saxena, 2019-04-06:
Did your simulation work after that? I get

```
creating netlist... done.
Starting /usr/local/bin/qucsdigi

running C++ conversion...

```
and after that 

`Segmentation fault (core dumped)`

bencarp, 2019-09-24:
Did you ever figure out how to solve the segmentation fault error? Feel like I've tried everything...

v2e, 2020-03-30:
> Did you ever figure out how to solve the segmentation fault error? Feel like I've tried everything...

I seem to have solved the issue by preparing and applying a set of patches which I attached to [this bugreport](https://bugs.launchpad.net/ubuntu/+source/freehdl/+bug/1781054/comments/7). If you use some Ubuntu-based distribution, you could try to build FreeHDL from the sources using my patches and build scripts.
I am not sure FreeHDL will work correctly after my "fixes", but at least I was able to perform some simple simulations in Qucs using my package.