Qucs/qucs #388: [OSX] App requires my password on install

alfuken, 2015-09-30:
Guys, really? Why would an app require my password except when it wants to install a rootkit?
I mean, normally on a mac no app should require user's password to get installed. If it does, then something fishy is going on and it tries to mess with the system. Could you, please, change the app to make the mac installation password-less, or at least describe a valid reason for doing so?
Thanks.

guitorri, 2015-09-30:
There are further issues... see also #131.

I wasted some time trying to get it to install without password. With no success. 

Indeed most Mac app bundles can be just dropped anywhere. However QUCS apps are not yet fully relocatable. Work is needed. 
We also need command line tools installed in `/usr/local`. I believe that by default you need elevated privileges to install stuff there. Even if you fancy building from source with Homebrew/MacPorts if I am not mistaken you have to `sudo` at some point to claim your `/usr/local` or `/opt/` to yourself...

Any help, expert advice about OSX packaging and patches are welcome. :wink:

felix-salfelder, 2015-09-30:
just don't install, if this is not applicable on your platform. run from builddir instead.

    mkdir myqucs;
    cd myqucs;
    $(qucstoplevel)/configure;
    make
    qucs/qucs/qucs
    cd ..; mv myqucs where/ever/you/want;
    where/ever/you/want/qucs/qucs/qucs

all you need to do is make this work for you. e.g.

    - finish #376 
    - hook `COND_MACOS` stuff into all-local, (now install-hook)

(note that i don't particularly care about osx)

guitorri, 2015-09-30:
@felix-salfelder keep in mind that Mac user (or Windows for that matter) in general do not build from source. 

Mac applications have a particular [bundle structure](https://developer.apple.com/library/mac/documentation/CoreFoundation/Conceptual/CFBundles/Introduction/Introduction.html). Libraries and resources are included into the bundle. In a way a bundled is a small sandbox, it should depend only on the stock OSX and the rest is provided in the bundle. If we want to distribute a binary version we have to conform somehow to this...

That is why we need fancy XCode SDK detection in the `configure.ac` files (to match the system libraries) and special `make install` targets to create the bundle structure. After that the software still has to packaged in a particular format... 

That is all for the convenience of our beloved end users :wink:

alfuken, 2015-09-30:
There's also a Homebrew (https://github.com/Homebrew/homebrew) , the package manager that can build stuff for you and keep it handy.
The main point is, though, you must **never** ask for a user's password or do anything that may require user to input his password.

felix-salfelder, 2015-09-30:
> @felix-salfelder keep in mind that Mac user (or Windows for that matter) in general do not build from source. 

easy. put the builddir into a tarball (after running make). maybe delete all the Makefiles before you do that.

> Mac applications have a particular bundle structure

i proposed to use the builddir as the root of such a bundle. if its not possible that would be certainly frustrating (and hard to believe).

guitorri, 2015-09-30:
@alfuken be my guest if you are Homebrew user: https://github.com/guitorri/homebrew-tap/blob/master/qucs.rb

@felix-salfelder, what I did was a `make install DESTDIR=/tmp/qucs/` and package that up (with the OSX `pkgbuild` and `productbuild` tools.

For some reason I cannot understand the result of [package creation] (https://github.com/Qucs/qucs/blob/master/qucs/contrib/pm/buildpkg_qucs.sh) only works with the root password... maybe I missed some option.

If anyone knows to fix that, please step up.

alfuken, 2015-09-30:
@guitorri you should list that at the download page - quite a lot of mac users use Homebrew.

guitorri, 2015-09-30:
Which page? It is listed in the website as a official package, it is also listed in the wiki. My homebrew-tap is the first hit on Google if you search for "qucs homebrew"... :smile:

alfuken, 2015-09-30:
@guitorri http://qucs.sourceforge.net

> Latest News
> 
> Latest stable release: 0.0.18
> Source 
> Windows Ubuntu OS X 
> Official SourceForge repository 
> GitHub Mirror 
> Qucs Wiki

And not a word about homebrew... ;)

felix-salfelder, 2020-08-31:
(why) is this a bug? looks like a feature request lacking a patch. can this be closed?

alfuken, 2020-08-31:
@felix-salfelder
1. homebrew still not mentioned anywhere. 
2. requiring a user's password is bad behaviour that should not be there in the first place. Call it a bug, call it a poorly implemented feature, call it whatever you like, just when closing don't forget to create new properly specified story, so it doesn't sunk forgotten.

Although seeing that previous comment was 5 years ago, and since then only a label been added... you know what... whatever.

felix-salfelder, 2020-08-31:
please adjust your attitude before reopening. and send a patch.

closed 2020-08-31 by felix-salfelder