Qucs/qucs #391: qucs schematic file extension

felix-salfelder, 2015-10-02:
qucs uses .sch as the file extension for what is called "schematic file". gschem uses the same extension.

qucs should be able to read/write file formats other than its own. verilog and gschem would be the first i'd like to implement. if this happens, the .sch extension will be ambiguous...

what shall we do about it? does it matter?

guitorri, 2015-10-02:
IMHO it does not matter. We should not rely on file extension for file processing.

felix-salfelder, 2015-10-02:
sure, but what will the File->Open dialog do about the types? (sorry, i am not used to graphical UI stuff, and this question might be stupid).

guitorri, 2015-10-02:
Perhaps have:
```
Qucs schematic (.sch)
gEDA schematic (.sch)
```
Ultimately the reader should check the contents and inform the user about unexpected file contents. No?

For the record KiCAD also uses .sch.

felix-salfelder, 2015-10-02:
> Ultimately the reader should check the contents and inform the user about unexpected file contents. No?

OK, the file selector simply lists *all* .sch files, and we need several readers, and a lookahead. should be feasible.

    reader = NULL;
    for (i in readers[".sch"]){
        if i->understands(file){
             reader = *i;
        }else{
        }
    }
    if(!reader){
        error("none of the readers can read your file");
    }else{
        try{
           reader->parse(schematic, file);
        }except e{
            error( reader->name() + "was not able to read file:" + e.string())
        }
    }

> For the record KiCAD also uses .sch.

ok, thanks for the input, all the more we will need the above.

closed 2015-10-02 by felix-salfelder