Qucs/qucs #394: The synthesis of parallel coupled transmission lines seems not to be working...

andresmmera, 2015-10-12:
Hello all, 

I am currently adding some new filter topologies to the Qucs-Filter tool ([branch](https://github.com/andresmmera/qucs/tree/NewFilters4QucsFilterTool)). However, I have realized that [this function](https://github.com/Qucs/qucs/blob/master/qucs/qucs-filter/tl_filter.cpp#L167)  is not working properly. Indeed, it is currently being used when synthesizing a microstrip parallel coupled filter, and it gives wrong results. Please, follow the steps below to reproduce this problem:

1) Open Qucs-Filter (Ctrl+2)
2) Select 'Coupled Microstrip' at the 'Realization' combobox and use the default substrate
3) Set 'Start frequency' to  30GHz
4) Set 'End frequency' to 30.5GHz
5) Generate the filter and paste it into Qucs.
6) After running the simulation, you will not appreciate a bandpass shape in S21, but if you spread the frequency sweep according to the picture below, you will see that the bandpass has been shifted to 33.5GHz. :-)

Microstrip lines:
[image]

Ideal coupled lines:
[image]

Is this issue already known? I can try to help in solving it, but it will take some time (I'm more short of time than usual).

In my case, I am trying to do a microstrip implementation of a side coupled ring filter [1]. It works fine as long as ideal coupled lines are concerned, but when doing the microstrip synthesis, it gives invalid values like (NaN, -1). Frustrated, I calculated the line parameters (length, width and gap) with the ADS Linecalc and I put them into Qucs. Not surprisingly, S11 and S21 were reasonable, and I concluded that something is wrong [here](https://github.com/andresmmera/qucs/blob/NewFilters4QucsFilterTool/qucs/qucs-filter/qw_coupled_ring_filter.cpp#L93)

Side coupled ring filter
Ideal lines:
[image]

Microstrip lines with ADS values:

[image]

References:
[1] Quarter Wavelength Side-Coupled Ring Resonator for Bandpass filters. Mohd Khairul Mohd Salleh, Gaëtan Prigent, Olivier Pigaglio, and Raymond Crampagne. IEEE TRANSACTIONS ON MICROWAVE THEORY AND TECHNIQUES. VOL. 56, NO. 1, JANUARY 2008

in3otd, 2015-10-13:
quickly, since I'm short on time too :grin: :

- coupled lines calculations code (and other transmission line code) is used in Qucs Transcalc, qucsator and qucsfilter, yet they **do not** share the same implementation. So if you compute the impedance with the qucs-filter code it's not sure that the qucsator code will compute exactly the same results. Yes, this does not make sense and should be fixed. A long, not-so-pleasant, error-prone task, so nobody has attempted to fix this, yet. If I compute the lines values manually using Qucs Transcalc the results seem more reasonable, AFAIU.
- in the first example the values are a bit "extreme", i.e. substrate H is comparable to wavelength and line spacing is much smaller than substrate H, not sure if the coupled lines formulas work well here. In general all the line synthesis/analysis code do not warn the user if values outside the allowable range are used, and this should be implemented.
- I cannot reproduce the results of the side-coupled ring filter, with ideal lines and  with the ADS values, please link the Qucs project, so I can be sure I'm simulating the same circuit.

andresmmera, 2015-10-13:
1) I should have tried Transcalc first...
2) Uhm, I see. I'll try a more suitable substrate.
3) [Here](https://gist.github.com/andresmmera/4e8f2438627942e3bb9e) is the schematic of the ring filter with ideal transmission lines. When 2) I'll upload the microstrip implementation
Thank you @in3otd :-)

andresmmera, 2015-10-13:
@in3otd Now, I've chosen FR4 as substrate (I've used it once @3GHz, so I hope it is ok for 1GHz). The substrate thickness, H = 0.508 mm, so it is significantly smaller than the wavelength (f=1GHz => lambda = 30cm). As you probably noticed, I'm not an expert on substrates, maybe I'm missing something important :-S

As far as the coupled lines component is concerned, Qucs Transcalc and ADS Linecalc give quite different dimensions and, consequently, S11 and S21 differ quite a lot. Below are attached the schematics:

a) [Ideal transmission lines](https://gist.github.com/andresmmera/4e8f2438627942e3bb9e)
b) [Microstrip lines + Transcalc](https://gist.github.com/andresmmera/c0466064fca2113ae060)
c) [Microstrip lines + ADS Linecalc](https://gist.github.com/andresmmera/d4c7f80cf6e2150cf473)

'Schematic b)' seems to be centred at 1GHz, but the upper side of the bandpass is (apparently) missing (I ignore why).
[image]

'Schematic c)' gives promising results since the shape of the bandpass is fine. Surely, the bandpass is shifted because the fact that the length of the coupled lines are shorter than lambda/4 (I've added an extra transmission line and the bandpass seems to go down to 1GHz). 
[image]

All this lead me to think that something is wrong with Qucs when coupled lines are involved...

closed 2019-06-12 by andresmmera