Qucs/qucs #397: Transient simulation freezes GUI

Lahorde, 2015-10-16:
Hi,

I was looking for a an electronic simulation tool and I tried qucs. I read wiki https://github.com/Qucs/qucs/wiki/Time-Domain-Simulation . When launching simulation GUI freezes. .sch file is here https://gist.github.com/Lahorde/8c01631b901515175e13 and logs are :

Simulation lancée le ven. 16. oct. 2015 à 15:17:33:024

création du fichier topologique… terminée.
project location: 
modules to load: 0
factorycreate.size() is 0
factorycreate has registered:
parsing netlist...
checking netlist...
netlist content
      1 Switch instances
      1 C instances
      1 L instances
      1 R instances
      1 Vdc instances
      1 TR instances
creating netlist...

I hope I missed something.

ra3xdh, 2015-10-17:
We know about switching circuit time-domain simulation bugs. Switches will not work in 0.0.18 for zero on-resistance. This is related to #34 . Switches simulation was fixed by @in3otd in 0.0.19 . You can set Ron>0 for switch and Qucs-0.0.18. I have tried with Ron=0.01 Ohm and your circuit was successfully simulated.

closed 2015-10-17 by ra3xdh

Lahorde, 2015-10-19:
Tested on 0.0.19. OK. Thanks for fix.

aksenofo, 2018-02-16:
On 0.0.19. Thanks for fix. Does not work.

in3otd, 2018-02-16:
um, some more details would be helpful. Which schematic does not work? The one linked in the first post seems no longer available.

aksenofo, 2018-02-16:
Looks like I was wrong.
I confused the freq ( 1G ) in '**AS voltage source**'.
It just took a long time to finish simulation. (  beginner :) )