Qucs/qucs #401: MOSFET library not working

JAndersM, 2015-11-28:
I get errors trying to use the included MOSFETs library
Either:
line 3: syntax error, unrecognized character: `<'
line 3: syntax error, unexpected InvalidCharacter, expecting $end
or
line 29: checker error, subcircuit type `MOSFETs_IRFZ24N' requires 3 nodes in `Sub:_net1', found 2
depending on chosen component in the library.

The standard MOSFET works and components from other libraries.

Quics 0.0.18, Ubuntu 14.04

ra3xdh, 2015-11-29:
Are you using left-side "Compoent Library" dock? This issue was fixed 0.0.19. This is duplicate for #39 and #292. The reason was that  dragging from library dock makes invalid netlist entry. This bug took place only for components that are represented as subcircuit. Discrete components are not affected this bug. 

Left-side component dock is buggy in 0.0.18. I recommend you to use `Tools->Component Library` instead of left-side dock to solve this issue with 0.0.18.

closed 2015-11-29 by ra3xdh

JAndersM, 2015-11-29:
Ok
Tack!

scorpdaddy, 2016-07-22:
v 0.0.19.  left-side "Component Library" dock still not working.  But neither is Tools->Component Library, produces "Cannot open "/c/qucs-win64-bin/share/qucs/.lib"  There is no name for the library file.

in3otd, 2016-07-22:
@scorpdaddy , this is a different issue, reported also in #542 .

A temporary fix, described there, is to set a variable in a command window and then start the program:

```
cd \qucs-win64-bin\bin
set QUCSDIR=c:\qucs-win64-bin
qucs
```

A fix is already in the current source code and will then be in the next release.

scorpdaddy, 2016-07-22:
Awesome!  Thanks!