Qucs/qucs #402: Why are there too much branches? What about merging and closing the not active ones?

timofonic, 2015-11-29:
Hello.

I seem confused by seeing too much branches out there about QUCS, both experimental and the master one (stable).

What about merging them and make navigation easier? What's the issue there?

Why not having "master", "devel", "experimental" and maybe some more but not much more?

Kind regards.

guitorri, 2016-01-24:
Valid points. We have been discussing this at #330.

At this moment I moved the following stalled branches by @bastien-roucaries to the SourceForge repository.
> use_std_vector
use_libeigen_v4
use_libeigen_v3
use_libeigen_2015_v2
use_hash_for_properties_list
simplify_further_2015-01-06
libeigen_2015
improvement_retry
improve_20150109

@ra3xdh are the 3 branches necessary for spice4qucs? What is the difference in them?

ra3xdh, 2016-01-24:
@guitorri  I can remove `spice4qucs` branch

`rebase_spice4qucs` branch is stable branch. It is intended for building Qucs-*S release candidates.
`spice4qucs_current` branch is for experimental features that could not be included yet in release candidates.

guitorri, 2016-01-27:
@ra3xdh I removed if for spice4qucs you.

I also remove some of my own stuff. The repo is a bit better now.

closed 2016-01-27 by guitorri

felix-salfelder, 2016-02-09:
it's a bit late, but now i can provide an answer to @timofonic's question. it's really a good question.

we have so many branches, because qucs is not modular. if qucs was modular, everyone could implement his own favourite stuff independently. without messing with the main git repository.

so let's make qucs more modular, and turn the random branches into modules/extensions/plugins.

and to answer the second question: no lets not merge. turn them into extensions.

nvdl, 2016-02-11:
+1 for modularity. But that seems as another project in itself.

felix-salfelder, 2016-02-11:
> But that seems as another project in itself.

it's not. see #464.