Qucs/qucs #403: Unable to compile qucs-core

andresmmera, 2015-12-02:
Hello all,

I'm having trouble compiling qucs-core. Specifically, it gives me this error when I run 'make':

./qucsator -l > qucsdefs.h
/bin/bash: line 1: 29518 Segmentation fault      (core dumped) ./qucsator -l > qucsdefs.h
make[4]: *** [qucsdefs.h] Error 139
make[4]: Leaving directory `/home/andresmmera/git/qucs/qucs-core/src'
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory `/home/andresmmera/git/qucs/qucs-core/src'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/home/andresmmera/git/qucs/qucs-core/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/andresmmera/git/qucs/qucs-core'
make: *** [all] Error 2

I have no clue about the origin of the problem, but, I did the following (without success...)

git checkout master
git reset --hard HEAD~10 //Reset my master branch to an older commit
git pull upstream master //Synchronize my master branch with Qucs' master branch

// Then, I simply followed the compile instructions at README.md
./bootstrap
cd qucs
./configure --enable-maintainer-mode
make
sudo make install
cd ../qucs-core
./configure --enable-maintainer-mode
//The problem comes here...
make
... and it shows the error I included above.

Am I missing something important? Any guess?

Thank you beforehand,
Regards

felix-salfelder, 2015-12-02:
> Any guess?

thanks for the report.

i'm not really sure, what `./qucsator -l` is supposed to do and why, but the binary seems to include several headers created by `admsXml`. our buildbot, travis, sets `--with-mkadms=admsXml`, after installing `adms` (that's also what I use to do). your attempt (probably) silently falls back to an adms built somewhere during the build process. I guess that your segfault is *not* related to this. I do not see anything more suspicious though. to be sure you might want to retry with preinstalled adms.

otherwise: can you give some more details on your segfault? maybe run the `./qucsator -l` thing in a gdb session... some information on your system or compiler might also help.

andresmmera, 2015-12-03:
Sincere thanks, @felix-salfelder 

After repeating (several times) the procedure I described before, I found that sometimes (rarely) it compiles fine... Besides this, I discovered that if I run the 'make' command twice, the second time it shows this:

In file included from check_spice.cpp:176:0:
../../src/qucsdefs.h:2749:5: warning: missing terminating " character [enabled by default]
   { "f2v
     ^
../../src/qucsdefs.h:2749:3: error: missing terminating " character
   { "f2v
   ^
../../src/qucsdefs.h:11:0: error: unterminated #ifndef
 #ifndef __QUCSDEFS_H__
 ^
check_spice.cpp:185:1: error: expected primary-expression before ‘struct’
 struct value_t * netlist_reverse_values (struct value_t * values) {
 ^
check_spice.cpp:185:1: error: expected ‘}’ before ‘struct’
check_spice.cpp:185:1: error: expected ‘}’ before ‘struct’
check_spice.cpp:185:1: error: expected ‘,’ or ‘;’ before ‘struct’

********************
LOTS OF WARNINGS (about unused structures)
********************

make[3]: *** [check_spice.o] Error 1
make[3]: Leaving directory `/home/andresmmera/git/qucs/qucs-core/src/converter'
make[2]: *** [install-recursive] Error 1
make[2]: Leaving directory `/home/andresmmera/git/qucs/qucs-core/src'
make[1]: *** [install] Error 2
make[1]: Leaving directory `/home/andresmmera/git/qucs/qucs-core/src'
make: *** [install-recursive] Error 1

Concerning my system, I'm using Ubuntu 14.04 LTS. I'll try to use gdb, but, to be honest, I've never used it before...

felix-salfelder, 2015-12-03:
> sometimes fine

hrm. looks like a race? cant find anything suspicious. can you do a

`$ make |& tee /tmp/qucsbuildlog`

and upload this log somewhere? (`|&` works in bash, dont know of other shells supporting this).

> ../../src/qucsdefs.h:11:0: error: unterminated #ifndef

the interrupted qucsdefs creation leaves behind a truncated header, hence the `#endif` (last line) is missing. (probably this truncation won't lead to the source of the problem, as some date might have been in the write buffer at the time of the segfault). let's try gdb. i will explain some basic commans, lets hope these are sufficient to locate the problem (or at least permit a guess...).

    $ gdb --args ./qucsator -l
    (gdb) run

this will output a lot of stuff (the header), and then crash. leaving behind a `(gdb)` prompt. now you can type `where`. it will print the stack during the crash. `up` will go up one call (sometimes the crash happens in a library, which is uninteresting, need to know what calls them). `down` gets you back. `list` lists the current code line (+ context) corresponding to the stack position.

now you can either post the whole stack trace (`where`) or try to find the last command in the qucsator code before the crash (`list`). this might help.

> Ubuntu 14.04 LTS

i do not have anything close to this, maybe somebody else...?

in3otd, 2015-12-03:
@andresmmera , try to clone the qucs repository to a new (temporary) directory and check if it compiles there. I have seen in the past too many times that some leftover files from a previous compile causes issues during a recompilation later.

I should have an Ubuntu 14.04 LTS VirtualBox that I used for some debugging in the past somewhere, I'll try to see if it's still working there, but don't hold your breath...

(FYI, as it happens too often with Github here I didn't receive any message for the issue, I just saw your and Felix's replies later...)

andresmmera, 2015-12-03:
@in3otd It seems you're right... I entirely removed the 'git/qucs/qucs-core' directory and synchronized again my local master branch with the Qucs master branch. So far, so good, no more segmentation faults...

Many thanks!

andresmmera, 2015-12-06:
Well, the compiling problems came back again... The make command worked properly for several times (until yesterday), but now it crashes every time I run it... I assure you that I didn't do anything dangerous, just trying to add a few fairly simple components to qucs-core.

Things I've tried (without success):
1) Remove entirely the git directory and clone the source code again following the [guidelines](https://github.com/Qucs/qucs/wiki/Contribution)
2) Clone the qucs source in another directory
3) Compile qucs-core without updating the sources. That is, I compiled it just after doing 'git reset --hard HEAD~10'

The compile log:
https://dl.dropboxusercontent.com/u/21294824/qucsbuildlog

I've also tried gdb. 'gdb --args ./qucsator -l' complains that it cannot find 'qucsator' (./qucsator: No such file or directory.), so I've tried 'gdb --args make' instead. After crashing, I've played with the commands you've cited above, but they do not provide significant information.

[Inferior 1 (process 22482) exited with code 02]
(gdb) down
No stack.
(gdb) where
No stack.
(gdb) list
1	dl-debug.c: No such file or directory.
(gdb) up
No stack.
(gdb) 

By the way, what operating system are you using for qucs development? I've considered moving to Fedora, but so far I haven't done it yet...

felix-salfelder, 2015-12-06:
sorry that should be `gdb --args .libs/qucsator`. make sure, you change to the directory, where make had a problem (`src`).

guitorri, 2015-12-08:
The 'qucsator -l' dumps a source file with data structs describing the registered components. This structs are compiled into 'qucsconv'. This data is used do convert netlists.

Could it be that your newly included component has an error somewhere that causes the crash?

andresmmera, 2015-12-08:
@guitorri Well, the thing is that it crashes even if I compile fresh code directly from the Qucs master branch. I've cloned it in a new directory (far apart from the directory where I usually work) and followed the steps below:

cd gittmp //Temporal directory for testing this issue
git clone https://github.com/Qucs/qucs
cd qucs/qucs-core
git submodule init //ADMS git submodule
git submodule update
./bootstrap
./configure --enable-maintainter-mode
make <- Crash!

The error log is the same as before. Here is [qucsdefs.h] (https://dl.dropboxusercontent.com/u/21294824/qucsdefs.h) If you take a look at the end of this file (lines 2542 to 2749) it seems that the definition of req_hicumL2V2p23 is not complete.

@felix-salfelder 

After doing what you've suggested, the 'list' command gives this:

(gdb) list
49    #include "module.h"
50   
51    #if HAVE_UNISTD_H
52    #include <unistd.h>
53    #endif
54   
55    using namespace qucs;
56   
57    /*! \todo replace environement name root by "/" in order to be filesystem compatible */
58    int main (int argc, char ** argv) {

the other commands say 'No stack'. I hope it helps...

in3otd, 2015-12-09:
@andresmmera , I have tried to compile qucs-core after cloning in a new directory, following the steps you listed above, in a VirtualBox with Ubuntu 14.04 LTS and it compiled fine.

Did you try to pass a different `prefix` to `configure` ? In the past I had sometimes to clean up the destination directory from previous executables and libraries to be able to compile without errors. Yes, this does not make sense, as it should not make a difference, but it did...

andresmmera, 2016-01-17:
Ops,... I've forgotten to close this thread. Somehow 'qucs/qucs-core/src/components/.dep/' and 'qucs/qucs-core/src/components/.libs/' changed their permissions so I couldn't remove them at user level. Since the name of these folders starts by '.', the Ubuntu file browser (Nautilus) don't show them (they are hidden folders)... This way, I couldn't realize that they still remained there... 
Besides this, as @in3otd suggested, I had to remove the qucs-core binaries (located at /usr/local/)

Thank you guys :-)

closed 2016-01-17 by andresmmera

guitorri, 2016-01-18:
Could it be that the makefile is running the `qucsator -l` from the path, instead of running from the build directory? That could explain why you had to remove the binary from `/usr/local/`

in3otd, 2016-01-18:
@guitorri, don't know if the problems I saw here locally had the same origin, but for sure it happened several times that I had to remove the `qucs` installation directory (which, for me is *not* in the path) to be able to recompile `qucs-core`...

andresmmera, 2016-01-18:
Uhm, as far as I understand, I had a buggy component which caused the makefile to crash. I reckon that before crashing, the makefile was creating "something" at 'qucs/qucs-core/src/components/.dep/' and 'qucs/qucs-core/src/components/.libs/' and the crash caused this "thing" to be corrupted. Since .dep and .libs folders changed their permissions and visibility, the 'corrupted stuff' remained there. Once I fixed the bug and cleaned those folders (.dep, .libs and the binary from /usr/local/), the problem disappeared.

I'm afraid I can't provide further details