Qucs/qucs #408: Release build scripta seem to be broken

ra3xdh, 2016-01-02:
After some recent changes on autotools build system release build scripts were not updated together and became broken. I have found this bug during preparation of my special release candidate wit SPICE features enabled. `master` branch is also affected. Two build scripts `contrib/release.sh` and `contrib/qucs-release.sh` don't operate properly. The following issues present:

* Qucs documentation (PDFs) cannot be compiled because LaTeX could not be invoked. Also verilog-A models are not converted to CPP code. Incomplete release package is build.
* If you replace `autogen.sh` by `bootstrap.sh` in these scripts, the first issue is gone. But resulting `*.tar.gz` package remain incomplete. If you use `release.sh` it contain root `configure` script but it builds only qucs-gui. If you use `qucs-release.sh` it doesn't contain root `configure` script. Such incomplete package could not be processed by `contrib/qucs-build-win32.sh` to obtain Win32 installer.

felix-salfelder, 2016-01-04:
according to the [documentation](https://www.gnu.org/software/automake/manual/html_node/Basics-of-Distribution.html), the `dist` target is meant to create distributable tarballs (such as release tarballs). the extra scripts that reimplement this seem to be obsolete.

in addition, there is a `distcheck` target, which additionally checks the tarball created by `make dist`. please see (merge) #387 to make `distcheck` work properly. merge #387 to include more tests.

tl;dr; if `make dist` creates an incomplete/broken tarball, then `make distcheck` will notice (and fail). no extra scripts needed.

guitorri, 2016-01-26:
I looking at the release and packaging scripts.

@ra3xdh at some point consider merging master into your branch. 

Right now all dependencies need to be in place to build the tarball. We might need to get used to that.

Previously admsXml, latex,..., were a bit of a MAINTAINER_MODE dependency. After the tarball was made they were not necessary for a `./configure && make install`.

The previous behavior was to include in the tarball the code produced by admsXml and PDFs from qucs-doc along with the rest of the source code (a few qucs-doc/tutorial/*.pdf are still being included).

When compiling the tarball:
* the previously generated .h and .cpp files (by admsXml) would be used for compilation.
* the PDF from qucs-doc were simply installed (not running latex)

Concerning admsXml IMHO it is fine to even remove it from the repository (it is a git submodule right now). It is  pretty easy to install it form the tarball. We can make binary packages for Windows later on. It also should possible to --disable-veriog-a-modules to skip the verilog-a models completely. 

Setting up the dependencies for qucs-doc might need further documentation. Alternatively we could provide a binary package containing the PDF files alone to complement a install with --disable-doc.

@Qucs/collaborators any comment?

felix-salfelder, 2016-01-27:
> Concerning admsXml IMHO it is fine to even remove it from the repository (it is a git submodule right now). It is pretty easy to install it form the tarball. We can make binary packages for Windows [...]

i think we can safely remove it (from the repo). we should remove the dependency as well, but i do not consider these tasks too urgent.

what's more urgent is the license situation. we must remove some of the verilog models to obtain a legally distributable tarball (whichever script we use to create this tarball). a release without these models is fine with me. an alternative would be, to make component models pluggable before the 0.0.19 release and ship a non-free extension to that. `--{en,dis}able-some-models` must be a run time option eventually.

> Setting up the dependencies for qucs-doc might need further documentation.

sounds good, maybe we should even add more checks for the build dependencies.

> Alternatively we could provide a binary package containing the PDF files alone to complement a install with --disable-doc.

let's wait until somebody asks for such a package (i fail to see a use case). `--disable-doc` is great fun, but it is not meant to be used by the ordinary (non-dev) user. if configure succeeds and docbuild fails, that's a bug.

ra3xdh, 2016-01-27:
@guitorri , I will rebase my branches soon as you suggest.

Now I have another problem with the new build system. I am trying to perform clean build of Qucs from the `master` branch. I have all dependencies installed and I am performing the following commands according updated README:

~~~
git clone --recursive https://github.com/Qucs/qucs
cd qucs
./bootstrap
~~~

And I obtain the following output from `bootstrap`:
~~~
$ ./bootstrap 
Running aclocal...
aclocal: couldn't open directory 'm4': No such file or directory
~~~
That's all. No `configure` is created in `qucs` and `qucs-core`. And I am unable to perform the next steps. I have tested Slackware-14.1 and Debian-7. What am I doing wrong?

in3otd, 2016-01-27:
...strange, on Ubuntu 14.04 the same is just a warning from `aclocal` , then the script continues as expected
```
Running aclocal...
aclocal: warning: couldn't open directory 'm4': No such file or directory
```
`aclocal` version is 1.14.1

May be related to [this] (https://lists.gnu.org/archive/html/automake-patches/2013-02/msg00040.html)

I think I tried with Slackware 14.1 also yesterday, but maybe I have an `m4`directory locally already.

guitorri, 2016-01-27:
@ra3xdh in all other distros we get an warning, not an blocking error, see this [build](https://travis-ci.org/Qucs/qucs/jobs/105020508#L1137) for example.

I am not sure which tool should create the m4 directory.
Creating the m4 directory should get you going. Please have a look. 
We can let the bootstrap script check for that and `mkdir m4` if necessary. 

I told you to consider  _merging_ master into your branch which is easier than _rebasing_ each time. See this tutorial [writeup](https://www.atlassian.com/git/tutorials/merging-vs-rebasing/summary) for further info. Merging is easier for you, and for others to track your public branch this way. You can always rebase later to cleanup unwanted commits prior to the final merge to master.

felix-salfelder, 2016-01-27:
> We can let the bootstrap script check for that and mkdir m4 if necessary. 

definetely. also, we should do something against the warning-noise in our ci build systems.

(sorry if that was my fault)

ra3xdh, 2016-01-27:
Yes, creating empty `m4` directory with `mkdir` helps. `qucs-gui` was successfully built. But `qucs-core` build terminates with error. 

Here is `configure` log for qucs-core: http://pastebin.com/cBBjX04d
Here is `make` log: http://pastebin.com/mLRsgwwc

I am using clean build, qucs `master` branch

ra3xdh, 2016-01-27:
I am using `gperf-3.0.4` and `automake-1.11.5`

felix-salfelder, 2016-01-27:
i can reproduce this.
`qucs/qucs-core $ AUTOMAKE=automake-1.11 ACLOCAL=aclocal-1.11 ./bootstrap`

results in a `qucs/qucs-core/Makefile.in` containing

    parse_citi.h: parse_citi.cpp
       @if test ! -f $@; then rm -f parse_citi.cpp; else :; fi
       @if test ! -f $@; then $(MAKE) $(AM_MAKEFLAGS) parse_citi.cpp; else :; fi

which is wrong. we need a rule for `parse_citi.hpp`.

i'll try to find a workaround... or can we just depend on recent autotools? it's just a development dependency, and you can install an arbitrary number of autotools versions in parallel (just in case you need an older one for some other project).

in3otd, 2016-01-27:
@ra3xdh ...it's Slackware which has a loong release cycle :grin: on my 14.1 I remember having updated `automake` and co. last year since also other packages could not be built properly, so I do not see this issue here. (BTW, it seems we will have 14.2 in a "short" while...)

felix-salfelder, 2016-01-27:
see #419

ra3xdh, 2016-01-28:
I have updated `automake` and the whole Qucs package was successfully built now without problems. This issue may be closed.

closed 2016-01-28 by guitorri