Qucs/qucs #409: 1K resistor is treated as 1 Ohm

denizzzka, 2016-01-06:
but 1k is 1000 Ohms

It is bug or feature? Spent hours for search related bug in my scheme

in3otd, 2016-01-06:
well, it's a feature (sort of) :grin:

when you write `1K` (note the upper-case), Qucs assumes that the `K` is the base unit (Kelvin), so it throws it away and keeps only the numerical value, `1`.
It does not check that the base unit is congruent, so you could also have a resistor with a value of `1 H` or `1 F` and it will always be considered as `1 ohm`

The metric prefix for the factor 1000 is `k` (lower-case) so if you write `1k` or `1 k`(or even `1 kK`) it's then correctly interpreted as `1000 ohm`.

...and yes, it will be nice if the unit used were checked to be the expected ones for the component, but this is not currently done...

Symbian9, 2016-01-17:
Is it a way to disable "Ohm" suffics and use only "1k" for "1 kOhm"(and "123" for "123 Ohm") for all resistors?
<img>http://radiolub.ru/uploads/2013/yx-1000a-1.jpg</img>

andresmmera, 2016-01-17:
The resistance unit 'Ohm' is added every time you instance a resistor, so you'll have to remove the word 'Ohm' manually. (See qucs/qucs/components/resistor.cpp, line 26).

in3otd, 2016-01-17:
as Andres said, the resistors have a default value of "50 Ohm" when instantiated from the components dock (note that it should be "50 ohm", but this is a long story...). Simple way is to copy around a resistor where you first modified the value string to remove the "Ohm" part, instead of instantiating it from the components dock every time. Assigning a value of "1 k" or "123" will of course work.
Once in the schematic, Qucs will not change the components values strings.

Symbian9, 2016-01-17:
Ok, thanks for info, how remove unit description. Think, in next realise it will be better change "Ohm" to "Ω" (and "kOhm" to "kΩ") as main unit symbol for resistance.

closed 2018-07-19 by guitorri