Qucs/qucs #410: IRC #qucs is empty

Symbian9, 2016-01-17:
As you sad on https://github.com/Qucs/qucs/wiki/Submitting-Issues page
<code># qucs IRC channel on chat.freenode.net are better places to get help.</code>
But for now this channel is empty. Did you register it?

andresmmera, 2016-01-17:
It seems that the IRC channel is usually empty... In any case, you can report issues or simply find help at qucs-help mailing list (http://sourceforge.net/p/qucs/mailman/qucs-help/)

in3otd, 2016-01-22:
...yep, I guess it's the usual "chicken and egg" problem; since the `#qucs` channel is rarely used people feel not encouraged to use it.

FYI, I have just joined after quite some time and was pleasantly surprised to find another use already there, so let's try to use #qucs more :grin:

guitorri, 2016-01-23:
I registered the #qucs and ##qucs channels at freenode back in Jan 2015. 
I will try to use it while I am doing something Qucs related.
If anyone wants to become an operator please let me know.

guitorri, 2016-01-27:
We are there... from time to time... I will try to put a topic on the channel.

closed 2016-01-27 by guitorri