Qucs/qucs #411: Noise parameter simulation broken

la3pna, 2016-01-19:
Noise parameter simulations under S-parameter simulations seems to be broken in 0.0.18 and taking the rest of the S-parameter simulations with it when the noise parameters are choosen. 

When setting noise parameter to yes, and doing a S-parameter simulation, the simulation fails with the following error message: 
"Errors occurred during simulation on ti 19. jan 2016 at 20:38:34:707
Aborted."
There are no messages in the "Errors and Warnings:" textbox. 

If doing the same simulation in 0.0.16 it gives a result that are reasonable, with no error. Have been able to re-create this problem with several models and simulations. All of the simulations that fails have been amplifiers with both DC simulations and S-parameter. 

A simple simulation of a attenuator does not show the same problem.

in3otd, 2016-01-19:
Hello,
I guess this could be the same issue fixed in #133 . Can you compile a new version from GitHub? Or for Windows you can download a snapshot [here] (http://sourceforge.net/projects/qucs/files/qucs-binary/0.0.19-snapshots/) . It's relatively old and might have some quirks but it contains the fix for the noise simulation issue above.

la3pna, 2016-01-19:
Yes, this solves the issue.  Thanks!
Closed.

closed 2016-01-19 by la3pna