Qucs/qucs #421: Remove from tarball non GPL verilog-a models

guitorri, 2016-01-27:
See #390 for context.

It is time for a release. 
The best we can do now is to somehow disable the verilog-a models from the build system and skip them from distribution.

Later on we figure out a way to either provide them in another format. Or if we are lucky, the authors of those modules will change they licenses.

guitorri, 2016-01-28:
Something like https://github.com/Qucs/qucs/compare/master...guitorri:remove-non-GPL-from-dist

felix-salfelder, 2016-01-28:
> Later on we figure out a way to either provide them in another format.

we must provide a way to use verilog models with qucs (in a straightforward way, without linking). then provide a package with verilog models under an appropriate license.

> Or if we are lucky, the authors of those modules will change they licenses.

even if this happens (i do not expect miracles), someone will implement another model that we might want to provide.

> Add the #define GPL (or another name) definition as build option.

licenses are mandatory, not optional. your idea is essentially to bypass the license. if this was legal, nobody would have to adhere to the GPL (in this case) very much. IANAL, but i can ask debian-legal about it...

guitorri, 2016-01-28:
You are right. :)
The dynamic loader can be improved to allow straightforward linking as you say.
Non-GPL stuff should *not* be in the repo.

felix-salfelder, 2016-01-28:
> You are right. :)

not in a way that is inherently obvious to me. (i tried to read the GPL once more).

> The dynamic loader can be improved to allow straightforward linking as you say.

i think we need to introduce some kind of (loadable) model library. (let's talk about that on saturday.)

> Non-GPL stuff should *not* be in the repo.

that would be nice. can you set up a qucs-models-nonfree repo, so we have a place to put them?

i think, what we *must* do, is remove all traces of "linking GPL code with non-GPL" code. that is, remove all references to GPL incompatible stuff (contained in GPLd source code) and in particular, all build rules that involve non-free sources...

guitorri, 2016-01-28:
I did not read the GPL again, but as I think now, the distributed tarball should reflect the repository 1-to-1 (ignoring the autogenerated stuff). Some distributions have ways to build packages directly from a repository. 

If we had a serious license infringement (think DMCA) they could shut our repository down.... 

We sill talk for sure, there are more issues that need to be tackled for a loadable models and libraries.

A non-free repo sounds right. We should keep the original file (when allowed) and the patches needed to process and load them.

Later today I will continue on removing the traces of non-GPL.

Should we [rewrite history]( http://programmers.stackexchange.com/questions/181260/should-i-rewrite-the-git-history-of-a-non-gpl-project-to-remove-gpl-code-that-wa) as well??

guitorri, 2016-01-28:
Please take a look at #425. I think I removed everything.

closed 2016-02-01 by guitorri

felix-salfelder, 2016-02-09:
> Should we rewrite history as well??

 a really good question. let's not do this.

that said: i have no clue about the consequences, but i hope you all agree with me.

guitorri, 2016-02-09:
The worst case scenario is someone forcing us to do that (unlikely). In that case we either carefully rewrite history (at the cost of every fork/copy out there becoming potentially headless), or the project fork itself from the point after the models where removed (throw away the whole history before that). 
I guess we leave it like that for the moment...