Qucs/qucs #431: Updates to CI services

guitorri, 2016-02-01:
Travis:
 - [ ] move to a container based instance
 - [ ] set to 2 cores
 - [ ] evaluate the need of branches for coverage update, doxygen update, unstable tarballs and ftp to SF
 - [ ] speedup OSX, create a homebrew cask for qt4 (including qt3support).
 - [ ] setup qucs-doc (latex and what not)

AppVeyor:
 - [ ] add --disable-dependency-tracking to speedup the build
 - [ ] enable qucs-test (install python and what not)

guitorri, 2016-02-09:
Added a few more tasks to AppVeyor.

felix-salfelder, 2016-02-09:
does it mean that travis actually supports "setup build environment once and just use it every time"?

(i was looking for something like it, but couldnt find)

guitorri, 2016-02-09:
you mean the container based thing? I think so, you can reuse the build env. After you create the container, they keep the image around, so it is pretty quick to re-start it. I think it is based (or can be used as) Docker. I quickly read about it, I don't know exactly what needs to be done to activate.