Qucs/qucs #434: Shall we deprecate the current qucs-help system?

guitorri, 2016-02-01:
Take a look at: 
https://github.com/Qucs/qucs-help
https://media.readthedocs.org/pdf/qucs-help/qucs-0.0.18/qucs-help.pdf
https://media.readthedocs.org/pdf/qucs-help-ru/qucs-0.0.18/qucs-help-ru.pdf

We can write better docs and support multiple languages with Sphinx.
We can distribute the HTML or PDF files. No need to maintain yet another application.

in3otd, 2016-02-02:
I agree in the medium(/short) term we can move away from the "integrated" qucshelp application; it is just a sort of limited browser, I don't think it currently provides more functionality than a standard browser, actually quite less. 
All the documentation need a refresh, I feel; we should also add some component-specific documentation, IIRC we discussed this briefly months ago and you prepared some automatic components list and properties extraction but cannot find anything on the Wiki at present (?)
As usual, we lack time and resources...

guitorri, 2016-02-02:
I also think the standard browser (+ PDF for printing) would be better.

You mean this (or something similar to this):
https://github.com/Qucs/qucs-manual
http://qucs.github.io/qucs-manual/0.0.19/html-en/index.html

We could add detailed component descriptions along with that.

in3otd, 2016-02-02:
yep, the [`qucs-manual`] (https://github.com/Qucs/qucs-manual) was what I had in mind, had almost forgot...
I was thinking we could have some (reStructured) text files containing a more detailed description for a component, it could be named after the component identifier (e.g. `Vpulse`, `.DC` or `TWIST`), which are unique IIRC and then the `qucs-manual` generation script could include this additional description, if present. Could this be done? Should I open an issue on `qucs-manual` :grin: ?

guitorri, 2017-01-30:
I updated https://github.com/Qucs/qucs-help and http://qucs-help.readthedocs.io/en/latest/.

Some figures are too large making the html version even larger. The pdf for each language is about 8MB (also due to the large images). Shall we distribute the pdfs only? What about keeping it strictly online (users can download what they want)?

ra3xdh, 2017-01-31:
> What about keeping it strictly online (users can download what they want)?

Yes, embedded help could be online. We can retarget `F1` key stroke to the system web browser that will open an URL at the readthedocs.org.

in3otd, 2017-02-06:
yep, and one day I'd really like to see something like #448, where pressing F1 with an open component dialog shows the manual page for that component... but I digress

guitorri, 2017-02-10:
#669 is ready for review.

ra3xdh, 2017-07-12:
Qucs-help was removed in #669 that is merged now. Closing.

PR #669 (2017-07-12): Remove qucs help
Remove qucs-help.
Add menu action to launch default browser at http://qucs-help.readthedocs.io/.

Closes #434.
closed 2017-07-12 by ra3xdh