Qucs/qucs #445: qucs-rescodes issues

guitorri, 2016-02-05:
- [ ] the menu is not extending
 - [ ] it does not changes the font, when changed in the Qucs settings.
 - [ ] the about dialog could state a correct reference to the license

[image]

in3otd, 2016-02-05:
yep, saw some time ago that this little tool was different, I think I may have already the first point done somewhere.
Regarding the "correct reference to the license", the text is the "standard" blurb used by the other tools, should we update all of them , then? I cannot find a reference for this exact text in the GPL, but googling it seems to be used in may other SW, e.g. https://www.gnu.org/software/help2man/

in3otd, 2016-02-06:
fixed the first two points, please advise about the third

guitorri, 2016-02-07:
hum... either nothing or something like:

```
Qucs-rescodes is part of the Qucs project.

Qucs is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2, or (at your option) any later version.

This software is distributed in the hope that it will be useful,but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more details..

You should have received a copy of the GNU General Public License along with Qucs, see the file COPYING. If not see http://www.gnu.org/licenses/ or write to the Free Software Foundation, Inc., 51 Franklin Street - Fifth Floor,Boston, MA 02110-1301, USA.
```

guitorri, 2016-02-07:
Or the stardard text from http://www.gnu.org/licenses/gpl-2.0.html

PR #449 (2016-02-07): qucs-rescodes update
Quick update for the `qucs-rescodes` tool to use a standard `QMainWindow` and menubar and the global Qucs font and language settings.
Fixes #445 .

Note that I have added a missing copyright/license header to a file, as it seems to me that it was simply forgotten by the original author.

(This little tool needs some refactoring/polishing, will be another PR, maybe)
closed 2016-02-07 by guitorri