Qucs/qucs #446: Change in Qucs Home requires restart.

guitorri, 2016-02-05:
When the user changes the "File > Application Settings > Locations > Qucs Home" a restart is required.
Can't we just close all the open projects/tabs and reload the list of projects from the new location?

guitorri, 2016-02-05:
Related: Pressing the '...' button to find the new Qucs Home launches a dialog that does not look native on Windows.

[image]

in3otd, 2016-02-06:
@crobarcro I was looking at the Application Settings dialog and noticed that the "Add Path with SubFolders" button actually adds *just* the subfolders not also the selected directory itself. Is this intended?
Seems a bit confusing to me, since the button just above, "Add Path", adds the selected directory (as expected)

crobarcro, 2016-02-06:
Yes, this is a bug, it should also add the directory

closed 2016-02-11 by guitorri