Qucs/qucs #450: Set windows icons in OS X ?

in3otd, 2016-02-06:
While looking at the code of `qucsrescodes` I saw it does not set the windows icon when `__APPLE__` is defined (see [here] (https://github.com/Qucs/qucs/blob/master/qucs/qucs-rescodes/main.cpp#L54-L57)), while other tools, like `qucstrans`, set it in every case (see [here] (https://github.com/Qucs/qucs/blob/master/qucs/qucs-transcalc/qucstrans.cpp#L257-L258)).
Does this make a difference? I do not have a Mac, so I have no idea...

guitorri, 2016-02-07:
There is no window icon on OSX. At best you get an icon on the dock. But for that the icon gets assigned when application is bundled... I don't think it makes any difference.

in3otd, 2016-02-10:
ok, so if you see no difference between e.g. `qucsrescodes` and `qucstrans` we can safely remove the `#ifndef __APPLE__` around `setWindowIcon()`

closed 2016-02-12 by in3otd