Qucs/qucs #453: Crash when changing background color of text document

in3otd, 2016-02-06:
When changing a text documents background color (using the Application Settings dialog) Qucs crashes.
To reproduce, open Qucs, create a new text document (File->New Text) and change the Application background color.

The problem comes from [here] (https://github.com/Qucs/qucs/blob/master/qucs/qucs/dialogs/qucssettingsdialog.cpp#L483), as now `TextDoc` is derived from `QPlainTextEdit`.

I'll put a fix together with the other stuff for #446 but feel free to push a separate hotfix if you would like to be sure that it will be in 0.0.19 :grin:

guitorri, 2016-02-07:
Confirmed on OSX. If you find a fix please inform.

in3otd, 2016-02-07:
...well, fix is just to use `inherits("QPlainTextEdit")` instead of `inherits("QTextEdit")`, as said it will be in the PR for #446, not sure I can finish it today.

Could you/someone please take a look if there is a similar bug related to the `inherits("QTextEdit")` in `simmessage.cpp` ? At a first glance it's about simulating VHDL/Verilog text files in Qucs, but I never did this, so I cannot check if it's a real issue.

BTW, there is a function [`QucsApp::isTextDocument`] (https://github.com/Qucs/qucs/blob/master/qucs/qucs/qucs.cpp#L2480-L2482) which is extensively used in `qucs.cpp` for this check, will be nice to reuse it around, so if we change the `TextDoc` inheritance another time just changing this function will be enough.
What will be the best way to do this? Declare it static? Or outside of a class, in `misc.cpp` ?

closed 2016-02-11 by guitorri