Qucs/qucs #455: Evaluate and update license status and headers

guitorri, 2016-02-07:
Is there an easy way to find out about missing license headers, copyright notices and potential issues in the codebase? I mean, how debian ftp masters scans packages for issues? Is this available somewhere?

What continue adding the author copyright on each file change? Or should we use Qucs Team and refer to the AUTHOR file? 

TODO:
 - [ ] update the license header in all files of the project, use the same template

in3otd, 2016-02-07:
oh, just looked at it this yesterday... I used the script `licensecheck` from the `devscripts` (2.15.10) Debian package.

A sample output running it over the entire Qucs repo is [here] (https://github.com/Qucs/qucs/files/120731/qucs_licenses.txt)

in3otd, 2016-02-07:
FYI, here is a good article on [Managing copyright information within a free software project
] (https://softwarefreedom.org/resources/2012/ManagingCopyrightInformation.html) .

In every case, I think we should update the copyright notices for the files we modified in the past years, to clearly state they were modified, as per the [GPL, section 2a] (http://www.gnu.org/licenses/old-licenses/gpl-2.0.en.html#section2)

A file header proposed some time ago, modeled after the [GPL Howto section] (http://www.gnu.org/licenses/old-licenses/gpl-2.0.en.html#howto) is [here] (https://gist.github.com/in3otd/da25223fe5f9414e1f9b)

felix-salfelder, 2016-02-09:
> Should we continue adding the author copyright on each file change?

sure. we should list authors per-file. an "author copyright" should not change unless a substantial contribution is merged. for (lots of) trivial changes/bugfixes, we might want to add a note "maintained by the QUCS team"...

my opinion. because this
(a) gives credit to the author(s) and
(b) is easy to understand and to do. and
(c) a bugfix is never done alone (review?), and  is unrelated to authorship.

YMMV