Qucs/qucs #463: Adjustments to Windows binary package

guitorri, 2016-02-11:
- [ ] enable multiple versions side by side
  - [ ] Let the .exe install by default in `[path]\Qucs-{version}`
  - [ ] create `Qucs-{version}` shourcut group

felix-salfelder, 2016-02-11:
> enable multiple versions side by side

been there, done that. gnucap-uf essentiially allows infinite parallel and independent instances installed (including extension package installs).

* it took quite a while to figure out
* it adds some build system complexity (comsiderably easier with autotools)
* i mostly used it to coinstall gnucap and gnucap-uf
* it is (really) hard to test, and will fall on our feet. guaranteed.
* it was a terrible idea

let me further explain the last item only: if i had used plugins from the beginning, i wouldn't have piled up an unholy mess (gnucap-uf), which i am now struggling to turn into plugins for upstream gnucap.

there's *one* proper way to cope with the problem you are working around.

* (for development) fix running qucs from builddir (#376).
* (for end users) allow to choose different sets of plugins to get different behaviour (#464).
* (to circumvent regressions) fix the regression, add a test case.

(imo this is entirely not a windows issue.)

guitorri, 2016-02-11:
I am not working around. To my knowledge it is just the way it works on Windows binary packages. 

In Windows (and Mac) applications often all needed libraries (non system ones) are shipped along with the executables. It is becomes sort of sandbox. Dependencies are resolved first inside the box, then outside in the system libraries. This avoids the so called "DLL hell" on windows. On Mac they have what is called a bundle. *A bundle is a directory with a standardized hierarchical structure that holds executable code and the resources used by that code.* One can also have plugins inside the bundle.

This issue is just about fixing the InnoSetup script to help the users of the installer to deploy two separated Qucs versions. They could also change the install path manually, but the icon group would still be overwritten. 

You might be right about plugins. But we don't have them yet. How is gnucap and its plugins managed on a Windows machine?

felix-salfelder, 2016-02-11:
(still, I don't understand why you need multiple versions side by side.)

> A bundle is a directory [..]

i would expect that these bundles are coinstallable by construction (i.e. i expect that they don't interfere with each other). what am I missing -- sure i *am* missing something.

> How is gnucap and its plugins managed on a Windows machine?

on a windows machine, the code of the following sort is activated (see `include/md.h`). the rest is the same.

    inline void* dlopen(const char* f, int)
    {
      return LoadLibrary(const_cast<char*>(f));
    }

i have never tried to run this, but others did. i assume that `LoadLibrary` knows about the bundle and prefers plugins within the same bundle...

guitorri, 2017-02-02:
We are not building the Windows installer at the moment, and no requests where made to make one. Closing for now.

closed 2017-02-02 by guitorri