Qucs/qucs #464: modularity, dlopen/dlclose

felix-salfelder, 2016-02-11:
we need to make functionality pluggable, before we tinker with it. whatever we add or change to the code should be done after cleaning up the current implementation by making it a plugin.

a plugin is a shared object that cen be added by using dlopen and removed with dlclose. what we need to make pluggable includes (amongst others)
* (collections of) symbols
* data sources and processing
* netlisters/schematic writers
* simulator drivers

plugins should be stored and maintained within external packages/git-repositories eventually, so it is easy to unlink them from the core project and development disentangles. however, as a starting point we should allow extensions to be part of the main project. I propose to put new plugins to recently implemented (and unstable) interfaces into `qucs/contrib/something` and include it into the build and test suite.

this thread meant to clarify the meaning of plugins and to explain the semantics and consequences. if you have questions or concerns, please ask. if you have a particular feeling that specific parts of the code cannot or should not be moved to a plugin, please open a new and dedicated issue.

felix-salfelder, 2016-02-12:
working on #458 i am confronted with the question of what to do about shared object extensions. different platforms apparently use different ones (e.g. so and dll?)...

this leads to some thoughts/questions
- should we hide extensions in the UI? (add them back behind the scenes...)
- maybe no extensions at all (don't like this one particularly...)
- should a file that makes use of "someplugin.so" also work on a non-so platform (using someplugin.whatever instead)?
- we could strip off extensions completely, or use an alternative extension. someplugin.qp?
- some weird magic (mangle/guess extensions where necessary?)

personally, i think stripping is evil. also an alternative extension will mask essential details. i prefer "hide in the UI" to "mangle". but i don't like "hide" too much
.
any thoughts?

in3otd, 2016-02-12:
@felix-salfelder from this side of the screen I do not clearly see what you have in mind.
I think I understand what the plugin are supposed to do, but I do not get why the related file names show up in the UI. Won't I see a nice description of what the plugin offers, instead?

Why is the extension stuff not transparent to the developer? I mean, compiling Qucs we get automatically an executable named `qucs` on Linux and `qucs.exe` on Windows. Am I too naive in expecting that the `.so` vs. `.dll` stuff is handled automatically, if needed?

felix-salfelder, 2016-02-12:
> from this side of the screen I do not clearly see what you have in mind.

if there are modules (see above) then there must be a way to load them.

>  but I do not get why the related file names show up in the UI

for the same reason, why `.sch` or `.dat` or `.va` (...) files show up in the UI. but these have a platform independent extension ...

> Won't I see a nice description of what the plugin offers, instead?

sure/easy (good idea) propose a common interface and implement it. gnucap has a "help" command/infrastructure, it could well be done similarly (or much simpler?).

> Why is the extension stuff not transparent to the developer?

it is. but we need to be prepared that also users might want to load extensions.

> Am I too naive in expecting that the .so vs. .dll stuff is handled automatically, if needed?

sure, it should be handled automatically. but somewhere in between "user wants to load a plugin" and "plugin is being loaded" there is something platform dependent going on. 

NB: i am currently starting to like the "hide in the UI" option. this is what all the linkers do...

in3otd, 2016-02-13:
hiding the file extension seems fine to me, especially if loading the is done thru a separate dialog, where you can load extensions only. It's similar to what used to be done in Qucs in the Project view on the left side. The `_prj` "extension" of the projects directories was not shown to the user. (Then a refactoring brought in some "technical limitations" and now you see all the projects directories ending in `_prj` and with the same "folder-like" icon.)

felix-salfelder, 2016-02-13:
> hiding the file extension seems fine to me

ok. i will add the extension down in the backend, just before calling `dlopen`. this is somewhat a strange thing to do -- thanks for your appreciation.

felix-salfelder, 2020-08-31:
this is in develop now, and made use of in the refactor branch.

closed 2020-08-31 by felix-salfelder