Qucs/qucs #467: Selecting wire label change font on diagram

guitorri, 2016-02-12:
When selecting a wire label (I believe it was the last label I placed in the schematic). The fonts on diagrams become boldfaced.

Select first label (as expected):
[image]

Select second label (fonts on table and other diagrams become bold):
[image]

in3otd, 2016-02-12:
this seems to come from [here] (https://github.com/Qucs/qucs/blob/master/qucs/qucs/wirelabel.cpp#L116); indeed saving the font before and restoring after, as done [for the other Elements] (https://github.com/Qucs/qucs/blob/master/qucs/qucs/components/component.cpp#L189-L283) seems to fix the issue. Probably the issue is there since 312d8cc9234cca9cf58e665b0b529bcca68c47f0 but did not have time to bisect.
Will check a bit more and do a PR later.

in3otd, 2016-02-12:
oh, just found another (small) bug related to the wire labels highlighting: label several wires with the label "A", several others with "B". If you select just one of the labels name (A or B) you will correctly see all the other labels with the same name highlighted (bold), If you now select also the other label (i.e. both A and B, with Ctrl-LMB) only the last label name clicked will be highlighted, the first one will remain selected but not highlighted.Looking at the code this is not the intended behavior, both labels names should be highlighted.

Besides, I do not remember having seen this nice highlighting stuff documented anywhere (true for many other things...)

in3otd, 2016-02-12:
another related bug: label a wire and also a node (not connected to the wire, of course). Click on a label and you'll see that the other one will *not* be highlighted. Yet the wire and the node will be connected in the netlist (as expected).

guitorri, 2016-02-16:
Fixed in release-0.0.19 (356a7fc6b4d231ff6a719dde72728b42a3c80a73)

closed 2016-02-16 by guitorri