Qucs/qucs #475: File format, standards, interoperability and EDA?

timofonic, 2016-02-14:
Hello.

Are there some kind of concern about file formats in the Qucs community? What about different available EDA software? Is the circuit simulation software considered usual to be part of "EDA package/suite" (shrinkwrap software? high end software?) idea or just in the common suites?

Does QUCS want to be exclusively a simulator or have more long term goals?

What about projects like gEDA and KiCad?

Kind regards.

ra3xdh, 2016-02-15:
From my point of view simulator should remain simulator. I explained my point of view here in some [discussion threads](https://github.com/Qucs/qucs/issues/193) earlier  and I repeat it.

Schematics for PCB design software contain several things that could not and should not be simulated (connectors, MCUs, FPGAs). It's not need to simulate the whole circuit. It will be more rational to split it into small blocks and simulate them separately. It will require less human and CPU resources. 

Adding of PCB design software format support will allow to simulate only very simple circuits from KiCAD, etc. But we will spent many time and resources to implement it. It other words it will allow to transfer from KiCAD only circuit with 3 transistors.  Look at such proprietary simulator as MicorCAP. It haven't such features. It's only simulator. But it is one of the best simulators. I think we should follow MicroCAP way.

The feature that will be useful will be integration with EM-simulators. Qucs was intended as RF circuit simulation tool, it have an excellent RF-design features. And EM-simulator integration will allow to extend RF circuit design capabilities further.

felix-salfelder, 2016-02-15:
> Are there some kind of concern about file formats in the Qucs community?

at FOSDEM, we have chatted with kicad and geda developers about common grounds for file formats and data exchange. it would be nice to get there in practic -- starting from symbols, over to schematic and layout formats that all projects will cope with....

in addition, many projects tend to go the "modular" way so in fact it should be a question of whether somebody is willing to add specific functionality, rather than whether is it going to be embraced by the respective upstream projects, or by partiucular developers. e.g. it would obviously be great to be able to simulate whole circuits, directly from the circuit board or chip model. we have all the pieces, they don't fit together yet.

guitorri, 2016-02-15:
(IMHO such open/broad questions are better suited to the qucs-devel mailing list)

@timofonic the answer is similar to the one you got at the kicad-developers list related their file format. 

We (the Qucs project and the FOSS EDA community) are concerned. So far there are no open standards agreed upon. Work is in progress. Shaping up the Qucs file formats is just one of the many pieces we need to sort out if we want to have interoperability.

As mentioned, a schematic capture for PCB is quite a different thing that schematic capture for simulation. Requirements and objectives are quite different. However it is entirely possible to move back and forth from a pcb-schematic a simulation-schematic. The tricky thing is to have a lossless round-trip exchange between different tools. Perhaps a lossy exchange would make more sense in the beginning.

In our context, one could go about having the Kicad schematic "skeletonized" and imported into Qucs. The connectivity is maintained (which is one of the most important things in a circuit). The primitives (transistors, resistors, capacitors, ...) can be directly mapped to an existing models (which in turn are mapped to an available simulator backend). The special components  (chips, connectors, ...) can be left as an empty box with ports to be filled in by the user. Maybe the user will swap the empty box by a SPICE model, or a custom behavioral model of some kind. IMHO is is really important to be able to move the network (the graph that represents a circuit) across the tools. This is also true the other way around. We could send a Qucs circuit to Kicad, and in Kicad the user does the part and footprint assignment, placement and routing. At least for a sup-circuits of a whole board this could work.

I don't expect to be able to run a full PCB simulation anytime soon, not without manufacturer's models (which are typically SPICE, PSPICE or similar). Don't forget microwave circuits. That would be nice if we could layout them somewhere. Depending on the frequency, you might also want to extract the parasitics of your PCB. So, simulation requirements are very specific to the kind of the solution you need... 

On the same token, assume someone designs (with common of the shelf components ) something like a power source, or an audio amplifier or a battery charger in Kicad. There is no reason why he/she should not be able to transfer that schematic to a simulation environment (that is the whole point of EDA). 

Bundling front-end (schematic capture, model builder) and back-end (simulation, solvers in general) in a EDA tool is fairly common. As mentioned, things should be modular (something we also need to improve).

IMHO Qucs should position itself in the simulation side of things. Which is already a very ambitious and long term goal. 

Long story short, the answer to all your questions is yes.

tipofthesowrd, 2016-02-16:
I follow  @guitorri on this.
Does anybody ever run a SPICe simulation a full complex circuit board with many different areas/functions?

I've never done it, at most we simulated parts of a design. Power stage, Audio stage, video stage, transmission line behaviour for broadcast or high speed, but not a single simulation for an entire design. That would require huge computer resources

I don't know about the IC design community though.

Since Qucs is already an excellent simulator (and now with spice4qucs) will hopefully get an even bigger following I think Qucs should concentrate on being the best FOSS simulator out there.
Since there are already some very well thought out RF features and benefits QUCS should continue to work on this imho since this is where it can make a difference. If you look at AWR MWO this RF simulator has three main components (or had, it has been some since i've used it professionally)

1) schematic editor for RF/microwave --> QUCS has this implemented quite nicely already. Mostly would require some work on more complex models (e.g. multiple coupled transmission lines) and perhaps some way to enhance the data displays and run equations on the fly from them

2) layout --> MWO lays out microstrip/stripline parts by pasting polygons on a screen and snapping them together. Qucs largely has this info already available from the schematic components and most of the work would be in defining exact measurements and exporting to DXF. 
IMHO this is a real added value because you take up the glove where KiCAD and other FOSS layout programs are struggling and have little interest in implementing this

2b) transfer to 3D em simulation. Well here's the tricky part. MWO does have an EM simulator but in the past used to interface well with better/other 3D em simulators by exporting said polygons. The user of the 3D EM simulator had to fill in details suchs as waveguide ports/dielectric/pcb thickness etc but the you didn't have to redraw your layout in the 3D em program and this was the big time saver. Perhaps if 2 is implemented 2b would not be so hard to interface with other EM FOSS such as openEMS.

3) system simulator: used for calculating link budgets, losses throughout an RF/circuits, intermodulation distortion, modulation parameters etc. --> Far outside of the reach of what QUCS wants/plans to do. No use in thinking about it now

Just my 2c

felix-salfelder, 2016-02-16:
lets do some myths debunking...

> Does anybody ever run a SPICe simulation a full complex circuit board with many different areas/functions?

this is a pretty common task. to a circuit simulator, a circuit board model looks like any other circuit.

> That would require huge computer resources

depends on the models that you use for the parts. simulating an operational amplifier can be much easier (faster) than simulating a single transistor...

> I don't know about the IC design community though.

visualize a bunch of people trying to cram a lot of stuff onto a flake of silicon. figure how they are locked down in a cage with bars made of proprietary software.

> Since Qucs is already an excellent simulator (and now with spice4qucs) will hopefully get an even bigger following I think.

qucs is not a simulator at all. qucs is a graphical frontend. in fact, there's also a simulator in the package, `qucsator`. it was meant for educational/research purposes, and it can do some noteworthy particular high frequency stuff.

> Qucs should concentrate on being the best FOSS simulator out there.

certainly not. there is a plethora of simulators out there already. even supporting all of them (which is one of the goals) will not turn qucs into a simulator. if it wasn't for gschem, qucs might become the best schematic editor one day...

more seriously, qucs has a chance to become the best graphical circuit analysis platform. let's focus on *that*.

nvdl, 2016-02-16:
> more seriously, qucs has a chance to become the best graphical circuit analysis platform. let's focus on that.

+1 for that and that is what brought me to the project.

in3otd, 2016-02-16:
yes, my experience is also that a complete board simulation is not what one wants/needs to do also.
Last board I did with KiCad was a small RF power amplifier with temperature-compensated bias. I had separate schematics/simulations for the DC bias and the RF part in Qucs. Having the two in the same schematic for simulation would have not provided any more insight.

Just saw your comment, @felix-salfelder ; IMHO the point was not whether a simulator could (in theory) simulate an entire board, but if having a direct translation from a full PCB layout to a schematic would be helpful. As said, most of the times the answer is "no, blindly doing this makes no sense". For some small boards it could still be useful, yet. Don't know if KiCad can export just a *part* of a circuit to a netlist; being able to import the netlist to a schematic in Qucs could be useful.
BTW, if someone could integrate [this] (https://sourceforge.net/projects/netlistviewer/) into Qucs, it will likely make many hobbyists happy.

Oh, and we may need to standardize some terminology; let's say that "Qucs" is the entire current suite of programs (qucs+qucsator+the various small and big tools) and "qucs" is just the GUI (we may need to rename it "qucsgui", one day)

Sure, `qucsator` is lacking in the time-domain simulations, but for RF work is pretty much the only FOSS option. Being able to use other (FOSS) simulators with the qucs GUI will allow one to choose the best tool for the job.

nvdl, 2016-02-16:
Why not try converting "NetlistViewer" to a schematic loader plugin?
Do we have plugin API?

tipofthesowrd, 2016-02-16:
> For some small boards it could still be useful, yet. Don't know if KiCad can export just a part of a circuit to a netlist

Indeed, I would not want to import/export an entire board. i'm responsible for a single domain (be it PSU, audio/video,...) I want to be sure my stuff works as expected and how I've explained it to the system architect. I trust my collegues who do the other stuff to do the same and to tell me what I can expect as input (or what they require as output).

Even the hacker/maker community where possibly a small team (or team of one) is working on a project would not need to export/import it all at once. Do you think for example the HackRF should be imported in a simulator as a single project to simulate it all?

> Don't know if KiCad can export just a part of a circuit to a netlist; being able to import the netlist to a schematic in Qucs could be useful.

You could copy part of a circuit to a new schematic, export to Qucs, add DC bias etc and off you go.

> 
Oh, and we may need to standardize some terminology; let's say that "Qucs" is the entire current suite of programs

I stand corrected, in this context I was talking about Qucs as a whole. Makes very little sense to me make the distinction since I think most of the users are using Qucs and qucsator and they also see it as one. (At least the people I know who use Qucs). That might change with upcoming spice4qucs

> Sure, qucsator is lacking in the time-domain simulations, but for RF work is pretty much the only FOSS option. Being able to use other (FOSS) simulators with the qucs GUI will allow one to choose the best tool for the job.

Hell yeah, and I'm looking forward to it for my small signal analog stuff (and even more if the data displays are improved a little more)

felix-salfelder, 2016-02-16:
> Why not try converting "NetlistViewer" to a schematic loader plugin?

there are lots of netlist->schematic converters out there. including commericial (sic!). turn the current .sch-reader frontend into a pugin and provide alternatives.

> Do we have plugin API in place?

work in progress. need some changes in the schematic reader to properly support symbol plugins. it might make sense to do it right on this attempt.

felix-salfelder, 2016-02-16:
> That might change with upcoming spice4qucs

no it has already changed with freeHDL and icarusVerilog. we need to disentangle the implementaion to get any further... (read: to not move backwards).

your help is appreciated.

in3otd, 2016-02-16:
@tipofthesowrd yes, "Qucs", "qucs" or "qucsator" have been practically the same thing for the users until now, it's not strictly true anymore with spice4qucs and with the more general support for external simulators (and possibly other stuff) felix is working on.

nvdl, 2016-02-16:
> turn the current .sch-reader frontend into a pugin and provide alternatives.

It was thinking to use "NetlistViewer" as a test plugin if API is in place.

felix-salfelder, 2016-02-16:
i see. there seems to be no file->import yet.

go for it `:D`. put your implementation into contrib/nl_import (or something like that). build a plugin. load through dlopen. (might be easier after my related PRs have been merged).

erichVK5, 2016-05-06:
Basic QUCS->gschem conversion is now possible with the following utility:

https://github.com/erichVK5/translate2geda/

To convert QUCS schematics successfully, you need to install the QUCS specific translate2geda gschem symbols into your gschem symbol library, so that the geometry of the network and connections are preserved during conversion, when gschem schematic symbols are inserted into the network.

felix-salfelder, 2016-05-06:
thanks @erichVK5. this is really good news.

now it makes even more sense to support geda symbols, and symbols in general, on the qucs side. with your symbols, this is no longer theoretical... with support for one type of symbols it will be *much* easier to support more types or even invent new formats (svg, xml, anyone?)

my take on this are PRs #383 (prepare to support symbols) #458 (loadable plugin support, such as a geda-symbol-support plugin). does anybody care to merge these into develop?

xtronics, 2016-05-10:
I agree with ra3xdh.
My concern is trying to be a PCB package as well will diffuse the focus and keep qucs from moving forward as a simulator. Keeping a narrow focus is more important than it might seem. 
 
Once upon a time I created some documentation for kicad, 
https://wiki.xtronics.com/index.php/Kicad  (for the record, I started off using mylar and tape - moved to CAD systems in 1986 - used many systems over the years ). 

There are a LOT of things that all have to work together - a HUGE project on it's own. Kicad consists of several GUI programs - Schematic capture, Library editor, PCB lay-out, Bitmap importer, PCB calculator, auto-routing, auto-placement etc etc .  I could imagine Qucs as part of the suite - possible sharing parts libraries - not sure even that would be worth it.  The tools I need in Kicad schematic capture would get in the way of the tools I need in Qucs - would make an insanely complex - unusable interface. 

There was someone trying to use the schematic capture of Kicad as a SPICE front end - wasn't really useful for me.  I'm really glad both projects exist. 

Trying to solve all the problems of a PCB CAD package would probably derail qucs from doing things like Monte Carlo etc etc..

erichVK5, 2016-05-12:
At the very least, if QUCS has a well defined and documented geometry for symbols, even if QUCS continues to paint them vs having a "symbol" file per se, it will facilitate efforts enabling the export of designed QUCS networks to schematic editors in PCB design toolchains.

I had to reverse engineer the origins and geometry of the symbols that QUCS uses. If there were a simple descriptor, perhaps contained in an individual "symbol" file, implementation of conversion tools would be simpler.