Qucs/qucs #476: Finalize release 0.0.19, todo list

guitorri, 2016-02-15:
- [ ] update NEWS.md
- [ ] create release candidate
- [ ] provide Windows binary package for freeHDL (other dependencies have packages)
- [ ] test optional packages 
- [ ] check issue related to verilog-a loader reported by Mike

---
Things still pending for the release.

ra3xdh, 2016-02-16:
> check issue related to verilog-a loader reported by Mike

@guitorri  , There was no blocking issue with Verilog-A. There was capacitor polarity issue in my Verilog-A synthesizer that will be not included in release. I will correct code in my branch soon. The `master` branch is not affected. You can mark this point as completed and proceed.

guitorri, 2016-02-16:
Hi @ra3xdh, I was testing Mike model against master. It was crashing on me while loading the .png (or dragging the component into the schematic). It was late, I will check again to see what I did.

guitorri, 2016-02-16:
@ra3xdh see #478 for the issue I saw.

felix-salfelder, 2016-02-18:
> provide Windows binary package for freeHDL (other dependencies have packages)

are you sure? i mean, is freeHDL unmaintained?

guitorri, 2016-02-18:
freeHDL is dead upstream.

guitorri, 2016-02-18:
by the way, we should deprecate freeHDL and switch to GDHL...

felix-salfelder, 2016-02-23:
> freeHDL is dead upstream.

do we really need it for the release? i mean, there's a lot of things to wait for, for example debian has no adms package yet...

> and switch to GDHL...

probably makes sense.

> [further up] test optional packages

which optional packages?

guitorri, 2016-03-13:
I will either create a Windows binary package for freeHDL or add it to the Qucs installer (as done in the past). The issue is that people use it as teaching tool. If we don't provide it, people won't be able to run it. Hopefully this will be the last time we package/ship freeHDL.

I crossed out the 'optional packages' (I had in mind icarus, freehdl, asco, octave, adms). It was a reminder to test these things manually just to make sure they are working as expected.

I am almost done with the other items. I hope to push a -rc1 tag someday this week.

guitorri, 2016-04-09:
I got a user report that the Verilog-A loader is not working on Windows. I am looking into it.

guitorri, 2016-11-04:
I will push the `-rc2` tag.

in3otd, 2016-11-06:
for the release, it would be good to update the [README.md](https://github.com/Qucs/qucs/blob/master/README.md) to explicitly tell that the `master` branch is the latest stable version, the bleeding edge is in `develop` and a potential release candidate would be in `release-x.y.z` so a use would know what ey will get (see also #570).

guitorri, 2016-11-09:
@in3otd can you check the wording in #597

guitorri, 2016-11-29:
So, `-rc3` was pushed 3 days ago. We had downloads:
 - 83 tarball
- 790 binaries (Win, Mac)
 
I did not hear any comments/complaints. 
I guess this is it. I will merge things around and prepare the release announcement.

To do:
- Merge `release-0.0.19` into `master` and `develop`. 
- Tag `qucs-0.0.19` into `master`. 
- Move the base of all PRs to `develop`
- Delete `release-0.0.19`

guitorri, 2016-11-29:
Crap, Travis changed something yesterday about Mac OSX images.... I will try to fix it else merge it anyway.

guitorri, 2016-11-29:
On top of that, the OSX images got stuck. Now it is back working, but there is a large backlog.... Release postponed till tomorrow.
https://www.traviscistatus.com/

guitorri, 2017-01-18:
Last pass. I will check #637 and do the release.

guitorri, 2017-01-22:
Release 0.0.19 has been tagged. 
I am working on the packages now.

guitorri, 2017-01-23:
Packages are staged (hidden from public) at Sourceforce till Wednesday 25 Jan.
Admins can take a look at them. @in3otd can you do a sanity check on the tarball and windows binary?
I might unstage all packages as soon finish spell-checking the announcement messages :smile: .

in3otd, 2017-01-23:
ah, this is why I didn't see them there yet :grin: I'll try to test them shortly...

in3otd, 2017-01-23:
uhm, logging in in Sourceforge I can see the `0.0.19`  directory in `qucs-binary` but I cannot download any file: all the files names are grayed out (they are not links) and an icon at the end says "pending". Clicking on the "*i*" button on the right I get a message "This file will be ready for download shortly."

guitorri, 2017-01-23:
The SourceForge staging thing is silly. It does not let an admin download the staged files.
I guess you have to uncheck the _Stage this folder_ in the `0.0.19` directory (in the "i" button), go inside, refresh the links, download what you want and stage the directory again. :man_facepalming:

in3otd, 2017-01-24:
testing with Windows, so far so good...

Seen only a (minor) ASCO issue that I sometimes saw also in Linux: when the optimization variables are constrained to be in a [E-series](https://en.wikipedia.org/wiki/Preferred_number#E-series), ASCO complains that the initial value (that ASCO itself chooses) is outside the allowed range, so the optimization is not started.
I remember looking at the ASCO code but the origin of this issue was not really clear to me; it also appeared/disappeared depending on the compiler optimization used.

The SourceForge staging is really weird, today the files names in the staged directory are links but when clicking on them I get an error message "*The file could not be found or is not available. Please select another file.*"... Did as you suggested, unstaged, downloaded and staged again.

guitorri, 2017-01-25:
Thank you. Please open an issue for ASCO. 
I will move on with the release.
Using the [FTP](https://sourceforge.net/p/forge/documentation/SFTP/) one can access the staged files on SourceForge.

guitorri, 2017-01-26:
Qucs 0.0.19 has been released.

closed 2017-01-26 by guitorri