Qucs/qucs #477: "NetlistViewer" as a Netlist Loader Plugin

nvdl, 2016-02-16:
[ ] - Wait for API to be ready

felix-salfelder, 2016-02-16:
what do you need? sure, the plugin stuff and the dispatchers (WIP).
can you add a file->importFrom dialog (with a dummy item). i can take care of how the plugins show up there.

i haven't had a look yet. probably you will go through

   Schematic::add_instance( something* ) 

what else?

nvdl, 2016-02-16:
Well, I will take it at a slower pace. Still, have other stuff on table.
Thus, I'll wait for the API to be ready while gaining momentum with this project again.

Are you documenting what is being done for the API so that I can get a feel of that?
Does the plugin depend on qucs code for compilation?

Any thought about making standalone binaries talking to qucs through a socket or something?
In that way, a developer just deploys that binary as a plugin that can be invoked from qucs or command line.

It will allows for other languages based (e.g. Python) plugins.

felix-salfelder, 2016-02-16:
> Still, have other stuff on table.

me, too...

> Thus, I'll wait for the API to be ready while gaining momentum with this project again.

there will be no api, before you tell me what you need in that API.

> Does the plugin depend on qucs code for compilation?

it will likely include a qucs header dedicated to this purpose.

> Any thought about making standalone binaries

see how the "netlister" plugins work, it's all trough `dlopen`.

> It will allows for other languages based (e.g. Python) plugins.

what you implement is completely up to you, open a new issue if you need something particular.

closed 2018-07-19 by guitorri