Qucs/qucs #478: Reloading verilog-a symbol/bitmap crashes application.

guitorri, 2016-02-16:
Steps to reproduce:

- `Project > Load Verilog-A module...`
- Pick and place from `Components > verilog-a user devices` (in the dock), should work normally
- (repeat) `Project > Load Verilog-A module`...
- Press `Change icon`
- Select another png icon, ok
- Pick and place from `Components > verilog-a user devices` --> CRASH

Traceback:

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qucs                      	0x000000010bcc53a0 QucsApp::slotSelectComponent(QListWidgetItem*) + 3440 (qucs.cpp:884)
1   org.qucs                      	0x000000010bd4f194 QucsApp::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 548 (moc_qucs.cxx:286)
2   QtCore                        	0x000000010d7dbb4e QMetaObject::activate(QObject*, QMetaObject const*, int, void**) + 1566
3   QtGui                         	0x000000010cf35e9e QListWidget::itemPressed(QListWidgetItem*) + 46
4   QtCore                        	0x000000010d7dbb4e QMetaObject::activate(QObject*, QMetaObject const*, int, void**) + 1566
5   QtGui                         	0x000000010cec2e56 QAbstractItemView::pressed(QModelIndex const&) + 38
6   QtGui                         	0x000000010cec78b9 QAbstractItemView::mousePressEvent(QMouseEvent*) + 1305
7   QtGui                         	0x000000010ca11940 QWidget::event(QEvent*) + 2816
8   QtGui                         	0x000000010cdaed5c QFrame::event(QEvent*) + 44
```  

Further issues:
- [ ] cannot unload symbols

guitorri, 2016-02-16:
This test project, by @MikeBrinson,  can be used to reproduce the issue [RC_prj.zip](https://github.com/Qucs/qucs/files/133353/RC_prj.zip)

ra3xdh, 2016-02-17:
Yes, I can reproduce this. Time-domain simulation issues reported by Mike are not caused by this bug. I have fixed Verilog-A synthesizer at my branch and it seems time-domain simulation issues are gone.

in3otd, 2016-03-11:
I have the impression that things get out of sync between the `Module::vaComponents`, which is cleared and reloaded [here](https://github.com/Qucs/qucs/blob/master/qucs/qucs/qucs_actions.cpp#L1383-L1418) and the `verilog-a user devices` Category Content, where apparently the Modules Content list is never cleared and new items are always appended ([here](https://github.com/Qucs/qucs/blob/master/qucs/qucs/module.cpp#L130)) even if the same component is added again.
So when you select the component in the Main Dock for the second time, its icon is at [index 0](https://github.com/Qucs/qucs/blob/master/qucs/qucs/qucs.cpp#L823) but the corresponding data are now at index 1.
Does this make sense? I'm not sure I understood how all this is supposed to work :confused: .

guitorri, 2016-03-12:
The crash should be fixed now. The code was using a non-mutable iterator to remove the modules. I guess the behavior is undefined. My understanding is that a selected (reloaded) component was pointing to a random location, crashing the code when an object was being called-back from that location.

Yep, the working principle is a bit unclear and poorly documented. I am not very proud of it. This should be reworked (along with a more generic framework for loadable libraries of components/symbols)...

MikeBrinson, 2016-03-13:
Hello Guilherme and Vadim
Thank you both for finding a solution to #478.
The problem effected the utility of the Verilog-A model development system whenreloading components/symbols.  The latest patch should make Qucs-0.0.19 muchmore robust.
Best wishes.
Mike
 Mike Brinson mbrin72043@yahoo.co.uk

closed 2016-03-16 by guitorri