Qucs/qucs #485: If transient simulation hangs.... auto-add parasitics?

arrgh123, 2016-02-23:
Hi, i tried out Qucs 0.0.18 as a new user. I really like it. The user interface is very intuitive and i got quite comfortable working with it within a few hours. 

I discovered though one pitfall for newbies: The transient simulation hangs... it hangs a lot. The culprit is most likely that there are some ideal circuit elements present in the schematic, like ideal switches, transistors without any parasitic capacitances etc... The simulator basically tries to find a solution with potentially infinitely short transients and hence selects the minimum time step. 

The workaround is adding small parasitic resistors in series to all circuit branches and small parasitic capacitors to all circuit nodes. I am currently doing this manually. I am used to work with Cadence ADE-XL, where i never experienced this issue. I think that the ADE-XL simulator auto-adds the necessary parasitics to ensure the solvability of the circuit before the simulation.

in3otd, 2016-02-23:
thanks for the feedback. It's true that the transient simulation is not as robust as it should. Some of the problems are/were due to bugs in the components models: a bug in the switch was fixed in the current development code, another fix for a bug in the transistor model should be integrated soon.
Other things remain currently unfixed, as the simulator getting stuck at (and actually even below) the minimum timestep.
As you said, often adding (small) parasitics helps convergence, but the simulator should do this automatically, if needed/requested. Making the simulator more robust is on the TODO list, but will take time.
Currently there is also some work ongoing to allow using other simulators within the Qucs GUI.

ra3xdh, 2016-02-24:
Yes, `qucsator` simulation kernel contains time-domain simulation bugs. These bugs were already reported #39 , #41, #186 and partially fixed by @in3otd . The reason of these bugs lies a much deeper than simple adding of necessary parasitic RC-elements. 

You can also use my experimental `0.0.19S` release candidates. It can use `ngspice` instead of `qucsator`. Ngspice shows the best results on switching circuits simulations. You can download the latest Qucs-0.0.19S-rc4 packages for two platforms here: https://github.com/ra3xdh/qucs/releases/tag/0.0.19S-rc4

closed 2020-09-26 by felix-salfelder