Qucs/qucs #492: cant compile qucs-0.0.19-160204-git-83cc216 on linux

gabber-tech, 2016-03-14:
Hi there,

trying to compile qucs 0.19 on kubuntu 14.04. 

./configure result: :)
` qucs-core version 0.0.19 () configured successfully.
`

make result: :(

/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I..  -I../src/math -I../src/components -I../src/components -I../src/interface -I.   -O2 -std=gnu++11 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT equation.lo -MD -MP -MF .deps/equation.Tpo -c -o equation.lo equation.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -I../src/math -I../src/components -I../src/components -I../src/interface -I. -O2 -std=gnu++11 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT equation.lo -MD -MP -MF .deps/equation.Tpo -c equation.cpp  -fPIC -DPIC -o .libs/equation.o
equation.cpp: In member function ‘int qucs::eqn::application::evalTypeFast()’:
equation.cpp:807:29: error: ‘gperfapphash’ has not been declared
     struct appindex * idx = gperfapphash::get (key, strlen (key));
                             ^
equation.cpp:811:48: error: invalid use of incomplete type ‘struct qucs::eqn::application::evalTypeFast()::appindex’
         application_t * app = &applications[idx->index];
                                                ^
equation.cpp:807:12: error: forward declaration of ‘struct qucs::eqn::application::evalTypeFast()::appindex’
     struct appindex * idx = gperfapphash::get (key, strlen (key));
            ^
make[5]: *** [equation.lo] Fehler 1
make[5]: Verzeichnis »/home/gabber/Tools/QUCS/qucs-0.0.19/qucs-core/src« wird verlassen
make[4]: *** [all-recursive] Fehler 1
make[4]: Verzeichnis »/home/gabber/Tools/QUCS/qucs-0.0.19/qucs-core/src« wird verlassen
make[3]: *** [all] Fehler 2
make[3]: Verzeichnis »/home/gabber/Tools/QUCS/qucs-0.0.19/qucs-core/src« wird verlassen
make[2]: *** [all-recursive] Fehler 1
make[2]: Verzeichnis »/home/gabber/Tools/QUCS/qucs-0.0.19/qucs-core« wird verlassen
make[1]: *** [all] Fehler 2
make[1]: Verzeichnis »/home/gabber/Tools/QUCS/qucs-0.0.19/qucs-core« wird verlassen
make: *** [all-recursive] Fehler 1

guitorri, 2016-03-16:
Can you tell us which version of gperf you have?

This reminds me of issue https://github.com/Qucs/qucs/issues/295

gabber-tech, 2016-03-23:
yes it seems to be:
```
gperf --version
GNU gperf 3.0.4
```
I am confused by the #295 which says it's fixed with #465 

Do I need to use a different version of gperf?

guitorri, 2016-03-23:
So far the issue was seen-and-fixed only for Windows. See b5f9ae60ae44
Maybe we should find a way for `gperfappgen` to always produce a CRLF.

in3otd, 2016-03-23:
I regularly build with `gperf` 3.0.4 with Linux (Slackware), without any issue - before and after the #465 patch. I have downloaded and compiled the sources from `qucs-0.0.19-160204-git-83cc216` successfully here, so I could not replicate the issue. Could you please, after trying to compile
- `cd` to the `qucs-core/src` directory
- type `./gperfappgen | cat -v` : you should see a plain text, without any strange `^M` symbols at the end of the lines
- type `./gperfappgen > gperfapphash.gph`
- type `gperf -I -m 8 gperfapphash.gph` : you should obtain a non-empty text file

to help understanding which step is failing

gabber-tech, 2016-03-23:
Got qucs-0.0.19-rc1 working on my Mint 17 machine. With gperf 3.0.4. 

I will try the input of in3otd when I am on the kubuntu machine again.

zerberros, 2016-05-19:
something similar for me:

```
libtool: compile:  clang++ -DHAVE_CONFIG_H -I. -I.. -I../src/math -I../src/components -I../src/components -I../src/interface -I. -O2 -std=gnu++11 -pipe -fno-exceptions -ldl -D__STRICT_ANSI__ -MT nodelist.lo -MD -MP -MF .deps/nodelist.Tpo -c nodelist.cpp  -fPIC -DPIC -o .libs/nodelist.o
clang-3.8: warning: -ldl: 'linker' input unused
In file included from nodelist.cpp:32:
./logging.h:37:8: error: unknown type name 'FILE'
extern FILE * file_status;
       ^
./logging.h:38:8: error: unknown type name 'FILE'
extern FILE * file_error;
       ^
2 errors generated.
Makefile:859: fallo en las instrucciones para el objetivo 'nodelist.lo'
make[5]: *** [nodelist.lo] Error 1
make[5]: se sale del directorio '/home/zerberros/source/qucs-release-0.0.19/qucs-core/src'
Makefile:905: fallo en las instrucciones para el objetivo 'all-recursive'
make[4]: *** [all-recursive] Error 1
make[4]: se sale del directorio '/home/zerberros/source/qucs-release-0.0.19/qucs-core/src'
Makefile:597: fallo en las instrucciones para el objetivo 'all'
make[3]: *** [all] Error 2
make[3]: se sale del directorio '/home/zerberros/source/qucs-release-0.0.19/qucs-core/src'
Makefile:768: fallo en las instrucciones para el objetivo 'all-recursive'
make[2]: *** [all-recursive] Error 1
make[2]: se sale del directorio '/home/zerberros/source/qucs-release-0.0.19/qucs-core'
Makefile:655: fallo en las instrucciones para el objetivo 'all'
make[1]: *** [all] Error 2
make[1]: se sale del directorio '/home/zerberros/source/qucs-release-0.0.19/qucs-core'
Makefile:380: fallo en las instrucciones para el objetivo 'all-recursive'
make: *** [all-recursive] Error 1

```

this error is on Antergos (Arch) at qucs-0.0.19-rc1

in3otd, 2016-07-02:
not sure if this was somehow due to #532, fixed in #534 .
I'm closing this since we could not reproduce the issue.
Please open a new issue if you still have problems in compiling `qucs-core`.

closed 2016-07-02 by in3otd