Qucs/qucs #495: qucs-doc fails to build out-of-tree

guitorri, 2016-03-20:
The `\@input{}` is failing to find the `../basics.tex` one directory up from `dcstatic.tex`

To reproduce:

```
$ cd qucs-doc/
$ mkdir tmp && cd tmp
$ ../bootstrap
$ ../configure
$ make
[... snip ...]
TEXINPUTS=$TEXINPUTS:.. /Library/TeX/texbin/pdflatex ../../../tutorial/dcstatic/dcstatic.tex
This is pdfTeX, Version 3.14159265-2.6-1.40.16 (TeX Live 2015) (preloaded format=pdflatex)
 restricted \write18 enabled.
entering extended mode
(../../../tutorial/dcstatic/dcstatic.tex
LaTeX2e <2015/01/01>
Babel <3.9l> and hyphenation patterns for 79 languages loaded.
(/usr/local/texlive/2015/texmf-dist/tex/latex/base/report.cls
Document Class: report 2014/09/29 v1.4h Standard LaTeX document class
(/usr/local/texlive/2015/texmf-dist/tex/latex/base/size12.clo))
\@input{basics.aux}
No file basics.tex.
\@input{titlepage.aux}
No file titlepage.tex.
! Undefined control sequence.
l.18 \tuttitle
              {
? 
```

felix-salfelder, 2016-03-20:
setting TEXINPUTS is certainly a problem, as it overwrites the (internal) TEXINPUTS search path and latex will fail to find files like `report.cls` immediately.

what we would *want* to do is set `EXTRA_TEXINPUTS=$(srcdir):$(srcdir)/..:./..`, however, there is no such variable. a possible workaround might be to set `TEXINPUTS=$(shell latex --tell-me-your-texinputs):$(srcdir):$(wheteverelse)`.

the closest approximation that i know of is `kpsewhich -var-value TEXINPUTS`, but i do not know whether this is portable, or even exists on non-texlive distributions...

kakini, 2016-06-14:
pls look at
http://tex.stackexchange.com/questions/93712/definition-of-the-texinputs-variable

"the trailing colon means "append the standard value of TEXINPUTS" 

TEXINPUTS=..: some_command

at least in texlive (Fedora 23) it seems to work.

felix-salfelder, 2016-06-14:
thank you @kakini.

this is working for me (#524). needs more work: nobody has cleaned up yet, after copying the (same, wrong) rules into 65 places...

felix-salfelder, 2016-08-17:
ok, i did some cleanup. i seems i cannot build the docs for some other reason

    /usr/bin/octave mszl.m
    error: 'eglobpar' undefined near line 23 column 1
    error: called from
        mszl.m at line 23 column 1
    Makefile:577: recipe for target 'mszl.eps' failed

has octave changed? (is it just me?)

in3otd, 2016-08-17:
FYI, in case your distribution does not provide a ready-made package - the official home page of epsTk seems not to be working, I have downloaded it from http://ftp.ubuntu.com/ubuntu/pool/universe/o/octave-epstk/octave-epstk_2.4.orig.tar.bz2 (and then expanded in `/usr/share/octave/site/m/`)

felix-salfelder, 2016-08-17:
thank you.

note (to self?): need to add `octave --eval eglobpar || echo please install octave-epstk` to configure.ac...

guitorri, 2016-10-30:
Fixed with #524.

closed 2016-10-30 by guitorri