Qucs/qucs #497: Wrong netlist for subcircuits with shorted ports

in3otd, 2016-03-23:
This may be an old bug: I have checked that it's present also in Qucs 0.0.16 and could be the same issue mentioned in the [BUGS file years ago] (https://github.com/Qucs/qucs/blob/master/qucs-core/BUGS#L66-L72).

When a subcircuit contains ports which are on the same net (shorted by a wire) the connections at top level are wrong. 
The simple circuit below has a subcircuit, SC1, which is just a short between the two ports
[image]
and as you can see, the simulation result is wrong.
Looking at the generated netlist
```
.Def:short_circuit _net0 _net0
.Def:End

Vdc:V1 in gnd U="1 V"
R:R1 out gnd R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
Sub:SC1 out in Type="short_circuit"
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
```
it's clear that the two ports are actually not connected in the netlist.

If the subcircuit shorts the ports via a 0 ohm resistor the results are correct.

The above test project is in this [netlist_bug.zip](https://github.com/Qucs/qucs/files/186766/netlist_bug.zip) file.

felix-salfelder, 2016-03-23:
what does "to withdraw the netlist" mean? did this involve a warning or error message back in 2007?

probably it's best to disallow more than one port per name -- cannot think of a use case...

in3otd, 2016-03-23:
not sure what the simulator did in 2007, at that time Qucs was at version 0.0.13 and I do not have this version installed.  May not be easy to compile from the sources now.

I saw this bug while trying to simulate a subcircuit with a resonant circuit to GND between the shorted ports - also in this case one of the subcircuit ports ends up unconnected.

felix-salfelder, 2016-03-23:
where *is* the bug?

is it the missing warning/error? or is is the subcircuit definition? where does it come from?

    .Def:short_circuit _net0 _net0
    .Def:End

(it looks a bit strange, doesn't it?) FWIW, a verilog component that does what you apparently expect, could look like the following. note the lack of port name collision or magic involved.

    module SC(a,b)
      [..]
      V(a,b)<+0.;
      [..]
    endmodule

or spice

    .subckt SC 1 2
    Vshort 1 2 0
    .ends

in3otd, 2016-03-24:
if the "short circuit" subcircuit actually has an RLC parallel circuit to ground the netlist generated by Qucs is
```
.Def:short_circuit _net0 _net0
C:C1 gnd _net0 C="1 pF" V=""
R:R2 gnd _net0 R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
L:L1 gnd _net0 L="1 nH" I=""
.Def:End

Vdc:V1 in gnd U="1 V"
R:R1 out gnd R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
Sub:SC1 out in Type="short_circuit"
```

then the flattened netlist build by `qucsator` is
```
  L:short_circuit.SC1.L1 gnd in L="1e-09H"
  R:short_circuit.SC1.R2 gnd in R="50Ohm" Temp="26.85" Tc1="0" Tc2="0" Tnom="26.85"
  C:short_circuit.SC1.C1 gnd in C="1e-12F"
  Vdc:V1 in gnd U="1V"
  R:R1 out gnd R="50Ohm" Temp="26.85" Tc1="0" Tc2="0" Tnom="26.85"
  DC:DC1 Temp="26.85" reltol="0.001" abstol="1e-12A" vntol="1e-06V" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
```

so, where is the error? Can be in both: the `qucs` subcircuit definition may be correct, it has two ports which are actually on the same net. Then `qucsator` fails to keep the ports connected when flattening the netlist. Or we can say that the ports of a subcircuit *must* be on different nets, then `qucs` should generate a different netlist for this particular case.

felix-salfelder, 2016-03-26:
> Or we can say that the ports of a subcircuit must be on different nets

exactly. this follows directly from "a port *defines* a net". we should adhere to that. then, the bug is in the netlister: it confuses nets and ports.

eventually, we want a net to be a first level object (="component") which has ports (with coordinates). this will facilitate netlisting and schematic exchange quite a lot: netlisting becomes a no-op, because the schematic already *is* a netlist. and schematic exchage will be simple because we could use verilog (a standard, surprise!) for that.

felix-salfelder, 2025-03-02:
> eventually, we want a net to be a first level object (="component") which has ports (with coordinates). this will facilitate
> netlisting and schematic exchange quite a lot: netlisting becomes a [...]

Took a little while to figure out. Turns out that

     .Def:short_circuit _net0 _net0
     .Def:End

is perfectly fine, and Qucsator simply does not support it. Exactly as Stefan noted on the bug list on 2007-12-30.

We now use Verilog and named connections. No more need to speculate what it means. We have

    module same_port (.a(i), .b(i));

listed and explained in IEEE1364-2005, Section 12.3.3 Port declarations. The bug in Qucsator is probably there to stay, but moving the issue.