Qucs/qucs #498: Fast zooming (Ctrl+Mouse wheels) doesn't work.

arhiv6, 2016-03-25:
In Qucs at fast zooming (using Ctrl + Mouse wheel) is not working properly. If you change the zoom is not fast - it works fine. But if you zoom in quickly, the scale is reset to the initial value - that is it does not work. Tested on Ubuntu15.10 32 and 64bits, version 0.0.18 and 0.0.19.
Video: [https://youtu.be/yOsCOftvIBM](https://youtu.be/yOsCOftvIBM)

in3otd, 2016-03-25:
I have tried to reproduce the issue (Slackware 32 b, Qucs 0.0.19), but I did not see this problem. Do you see a similar behavior also with panning (just the mouse wheel and Shift+Mouse wheel)?

arhiv6, 2016-03-25:
No, Mouse wheel and Shift+Mouse wheel working normal.

andresmmera, 2016-03-25:
I've noticed that behaviour some time ago... but in the end I can live with it so I've never reported it :-)
Here i'm using Ubuntu 14.04 LTS 64b

in3otd, 2016-03-25:
ok... can you try to insert a line 
```
qDebug() << delta;
```
[here](https://github.com/Qucs/qucs/blob/master/qucs/qucs/schematic.cpp#L1882) and recompile, looking then at the numbers printed on the terminal while zooming, just to check if some strange step value is passed to the routine. I always get just `60` or `-60`.

andresmmera, 2016-03-25:
Uhm,... I've followed your instructions and after doing some random fast zooming (CTRL+Wheel) I can see many 60 and -60, but also 120, -120 and -180

andresmmera, 2016-03-25:
I can confirm that it shows (-60, 60) when you scroll slowly. However, as soon as you scroll faster, the effect described above arises and the log shows some 120 and 180.

andresmmera, 2016-03-25:
I've reduced the zooming step by replacing  "int delta = Event->delta() >> 1; " by   "double delta = Event->delta()*0.7;"  at line 1881 and it seems to work here...

in3otd, 2016-03-25:
thanks; could you please try changing the lines [here](https://github.com/Qucs/qucs/blob/master/qucs/qucs/schematic.cpp#L1894-L1895) to
```C++
if(delta < 0) Scaling = 1/1.1;
else Scaling = 1.1;
```
and check whether the zoom issue is gone

andresmmera, 2016-03-25:
Yep, this way the problem disappeared too :-)

in3otd, 2016-03-25:
just saw your new comment: looking at the code I think that the multiples of 60 returned by the wheel event are causing the zoom to actually reverse direction. Just taking the sign as suggested above should be enough, I don't think a fast(er) zoom is really needed in practice.

andresmmera, 2016-03-25:
Of course, a faster zoom is not needed at all. It was kinda quick 'patch'...

in3otd, 2016-03-26:
well, I think we can keep the faster zoom feature at little cost and this will also accomodate any mouse returning *less* than 120 for the wheel `delta()`. Please try #499 and check how it works there.

guitorri, 2016-11-03:
The fix is merged in both, develop and release-0.0.19.

closed 2016-11-03 by guitorri