Qucs/qucs #502: solved issue, developers attention needed

alysogorov, 2016-03-30:
Hi. I solved an annoying(at least for me)  issue with wrong cross painting in wire creating mode of schematics editor.
It is function  
 - file mouseactions.cpp
void MouseActions::MMoveWire1(Schematic *Doc, QMouseEvent *Event) where coordinates of crossed lines were calculated wrong.
Here is correct code of this function( line 346 of the file).
old code is commented in /*...*/ section
void MouseActions::MMoveWire1(Schematic *Doc, QMouseEvent *Event)
{
  /*
	MAx3 = DOC_X_POS(Event->pos().x());
	MAy3 = DOC_Y_POS(Event->pos().y());
	Doc->setOnGrid(MAx3, MAy3);

	MAx2  = DOC_X_POS(Doc->viewport()->width());
	MAy2  = DOC_Y_POS(Doc->viewport()->height());

	Doc->PostPaintEvent (_Line, Doc->ViewX1, MAy3, MAx2, MAy3);
	Doc->PostPaintEvent (_Line, MAx3, Doc->ViewY1, MAx3, MAy2);
	Doc->viewport()->update();
*/

	//we are working on local variables and will assign values
	//to object properties at the end
	int lx = DOC_X_POS(Event->pos().x());//get mice coordinates translated to document space
	int ly = DOC_Y_POS(Event->pos().y());
	Doc->setOnGrid(lx,ly); //align on grid

	//let we reserve couple of points at the edges of lines for some aesthetics,
	//and visual check that our calculations has fit the widget window.
	const int ldelta = 2;

	//left and upper edges of our lines
	int lx0 = DOC_X_POS(Doc->contentsX()+ldelta);
	int ly0 = DOC_Y_POS(Doc->contentsY()+ldelta);

	//right and bottom edges
	int lx1 = DOC_X_POS(Doc->contentsX()+Doc->viewport()->width()-1-ldelta);
	int ly1 = DOC_Y_POS(Doc->contentsY()+Doc->viewport()->height()-1-ldelta);
	
	Doc->PostPaintEvent (_Line, lx0, ly, lx1, ly);
	Doc->PostPaintEvent (_Line, lx, ly0, lx, ly1);
	//simulate behaviour of old code with object variables.
	MAx3 = lx;
	MAy3 = ly;

	//why it is needed i do not know, just for equivalence  :)
	MAx2 = lx1;
	MAy2 = ly1;

	Doc->viewport()->update(); // seems this call is excessive??? but let it be.
}

Solution is tested, and works correctly.
Someone has to put in into git(I'm a plain user and have not access).

Regards
Alex.

ps. day ago I've put the same letter to qucs-devel@lists.sourceforge.net, was recommended to put the letter here, but here I added a couple of lines and comments to the code, so this variant is more complete than from my letter to qucs-devel@lists.sourceforge.net

guitorri, 2016-03-31:
Thank you for the fix. I fixed the C++ formatting for you. If you see yourself doing more fixes, please consider sending a pull request (or a patch). There are some directions on our wiki about [contribution](https://github.com/Qucs/qucs/wiki/Contribution) via GitHub.

alysogorov, 2016-04-02:
Guilherme, when I created my local branch to fix the bug, i forgot to check, which git branch i have as base, and now it is master. So i ll do pull request to it. Not, for example, "develop" branch.
Is it good? Or i have to change base branch?

felix-salfelder, 2016-04-03:
it does not matter much (develop would have been better). you can change this any time.

theres this catch: when you file a PR, you are asked about which branch you want to merge your branch into. the default setting is `master`. the correct setting is `develop`. github does not allow to change this afterwards (and a million of users have complained about it...).

alysogorov, 2016-04-03:
Hi felix...
But what is the most correct way?

Because i made corrections only in one file, i saved it locally and can 
insert it in any my branch...
but what I have to do?

I see next variants
1. I get master branch to my computer, create local branch to fix 
issue(kinda branch A), make changes, commit to A, push to my github rep, 
ask to request to "develop" branch... it look quite strange for me, 
because bases are different - master and develop.
2. i just get "develop" branch from github, do the same(as above). So 
bases are equal... it looks better.
3. moreover some people are calling PR to branch 0.19....wtf really?

all looks like complete disorder... is there any instruction, which 
branch is for what?

i have not big experience with git,  at my work, there is special guy 
who joins different branches to master and it is his responsibility to 
keep project stable.

sorry if i am too annoying :)
bit it is my first business with github, and open source.

Alex.

alysogorov, 2016-04-03:
People, seems question solved in my email conversations with Vadim.
Thanks.

felix-salfelder, 2016-04-03:
@alysogorov 2 looks good. 3 is if you have a hotfix for the upcoming 0.0.19 release.

there is no special guy for this, so everyone must bring some common sense. we need to agree on some basic rules. cf. #330 for some more background.

mirandadam, 2016-10-20:
Hi! Can this issue be closed?

alysogorov, 2016-10-20:
if my code somehow went to code of Qucs, then yes. imo.

felix-salfelder, 2016-10-22:
(sorry, i'm not fully aware what this is about).
is there a pull request with a patch anywhere?

in3otd, 2016-10-22:
I think #504 is the corresponding PR

alysogorov, 2016-10-22:
> I think #504 is the corresponding PR

yes, it is.

guitorri, 2016-11-03:
Fix merged.

closed 2016-11-03 by guitorri