Qucs/qucs #506: Handle removed Verilog-A components when loading an schematic.

guitorri, 2016-04-09:
Schematics containing any of the removed Verilog-A components cannot be opened. The components are unknown...

The dialog below helps, but perhaps we should add a button to skip the missing component. So user can try to recover the schematic (use another component, of the Verilog-A loader ). 

[image]

felix-salfelder, 2016-04-09:
there should be a "yes" and a "no" button instead of the "OK" one.

joke aside. shouldn't is be possible to make the symbols available for the transition? the actual problem is that qucsator does not know what a `bsim4v30nMOS` is. *that* should be reported as an error...

ra3xdh, 2016-04-14:
See #507 for fix.

PR #507 (2016-10-31): Enabled loading of unknown components
This fixes #506. If Qucs find unknown component, a new dialog window with Yes/No buttons will be shown. User can press Yes and load schematic anyway, and press No and abort loading of schematic. Unknown components will be replaced by placeholders. I am using Subcircuits as placeholders. Maybe it's better to create some special placeholder device, which will be available for this purpose only? Should this PR will be for `release-0.0.19`?
closed 2016-10-31 by guitorri