Qucs/qucs #511: Still doesn't uninstall on OSX

olr666, 2016-05-10:
Hi, I know I made a post about this a long time ago but I got caught up in exams and forgot that QUCs was on my system, I'm running Mac OSX 10.11.4

I've just noticed it is still there and the instructions in the original response to me at https://github.com/Qucs/qucs/issues/348 as well as the uninstall script file downloaded directly both do not work.

When using the directions in the reply to my last thread on Github I get the following error

./uninstall-qucs-pkg.sh: line 1: Not: command not found

When using the uninstall script file I get the following error

sudo: /Users/My_User_Name/Downloads/uninstall-qucs-pkg.sh: command not found

Does anyone have any idea how to uninstall the software from my Mac?

guitorri, 2016-05-26:
Can you try to open the `uninstall-qucs-pkg.sh` in a text editor to see if there were no issues during download.

olr666, 2016-07-05:
Sorry for the long delay - I am studying and keep getting caught up in my final year project.

I had opened it in the text editor before posting here in May, and it seemed to have downloaded properly. I didn't have that file on my computer, so I just downloaded a new version from here https://sourceforge.net/projects/qucs/files/qucs-binary/0.0.18-snapshots/uninstall-qucs-pkg.sh/download

I attempted to run this file, and again got the same error of "command not found"

It appears to have downloaded correctly, here is the text

#!/bin/sh
(( ${#} > 0 )) || {
  echo ''
  echo '*** Qucs uninstaller script for Machintosh ***'
  echo ''
  echo 'DISCLAIMER: USE THIS SCRIPT AT YOUR OWN RISK!'
  echo 'THE AUTHOR TAKES NO RESPONSIBILITY FOR THE RESULTS OF THIS SCRIPT.'
  echo "Disclaimer aside, this worked for the author, for what that's worth."
  echo 'Press Control-C to quit now.'
  echo 'You may need to use as sudo.'
  read
  echo 'Re-running the script with sudo.'
  echo 'You may be prompted for a password.'
  sudo ${0} sudo
  exit $?
}

# bill of materials
packages=(
/var/db/receipts/org.qucs.pkg.bom
)

# clear files from packages
for bom in "${packages[@]}"; do
  if [ ! -e ${bom} ]; then
    echo "bom file not found: ${bom}"
  else
    #echo "found ${bom}"
    lsbom -f -l -s -pf ${bom} \
    | while read i; do
      # Remove each file listed in the bom.
      file="/${i}"
      if [ -e $file ]; then
        echo "removing: ${file}"
        rm $file
      else
        echo "file not found: ${file}"
      fi
    done
  fi
done

echo "*** Removing receipts: /var/db/receipts/org.qucs.*"
rm -f /var/db/receipts/org.qucs.*

olr666, 2016-07-05:
I also just retried your directions in #348, and I still receive the command not found error when I try to execute the third command you told me to, being:

`./uninstall-qucs-pkg.sh
`

In this case, the file has not downloaded correctly and the contents of the uninstall-qucs-pkg.sh file are as follows

`404: Not Found
`

olr666, 2016-07-17:
Managed to resolve this issue, copied the text from here 

https://sourceforge.net/p/qucs/git/ci/master/tree/contrib/mac/uninstall-qucs-pkg.sh

into the .sh file that wasn't downloading properly in the directions given in #348. Then I followed the rest of the steps and it worked.

Thanks!

in3otd, 2016-07-17:
Glad to see that it worked in the end - I'm closing this issue then. 
Feel free to open a new issue in case you have further questions.

As you noticed, the OS X uninstall script has been moved, it's now at
https://sourceforge.net/p/qucs/git/ci/master/tree/contrib/mac/uninstall-qucs-pkg.sh?format=raw
or equivalently at
https://raw.githubusercontent.com/Qucs/qucs/master/contrib/mac/uninstall-qucs-pkg.sh

closed 2016-07-17 by in3otd