Qucs/qucs #512: unable to compile qucs-core

stonecome, 2016-05-22:
Hi,all
I am meeting an error when I compile qucs-0.0.19(master branch) according to the README file. It stops in the qucs-core compiliation. Then I make the qucs-core after commands(./bootstrap ./configure), it reports:
Projects/qucs/qucs-core# make
make  all-recursive
make[1]: Entering directory `/home/leiyc/Projects/qucs/qucs-core'
Making all in src
make[2]: Entering directory `/home/leiyc/Projects/qucs/qucs-core/src'
g++ -DHAVE_CONFIG_H -I. -I..  -I../src/math -I../src/components -I../src/components -I../src/interface -I.   -O2 -std=gnu++11 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT gperfappgen.o -MD -MP -MF .deps/gperfappgen.Tpo -c -o gperfappgen.o gperfappgen.cpp
mv -f .deps/gperfappgen.Tpo .deps/gperfappgen.Po
/bin/bash ../libtool --tag=CXX   --mode=link g++  -O2 -std=gnu++11 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic   -o gperfappgen gperfappgen.o  -lm 
libtool: link: g++ -O2 -std=gnu++11 -pipe -fno-exceptions -fno-check-new -rdynamic -o gperfappgen gperfappgen.o  -ldl -lm
./gperfappgen  > gperfapphash.gph
/usr/bin/gperf -I -m 8 gperfapphash.gph | sed -e 's/{""},/{"",0},/g' > gperfapphash.cpp
make[2]: *** No rule to make target `parse_citi.hpp', needed by `all'.  Stop.
make[2]: Leaving directory `/home/leiyc/Projects/qucs/qucs-core/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/leiyc/Projects/qucs/qucs-core'
make: *** [all] Error 2
The environment info of my compilation is that:
Debian 3.2.68-1+deb7u2 i686 GNU/Linux
gperf 3.0.3
automake (GNU automake) 1.10.3
gcc (Debian 4.7.2-5) 4.7.2
bison (GNU Bison) 2.5 
lex 2.5.35...
Am I missing something important? Any guess?

Thanks in advance
Best Regard

guitorri, 2016-05-26:
Seems related to #295 and #492.

stonecome, 2016-05-26:
I check that  the output of command "/usr/bin/gperf -I -m 8 gperfapphash.gph | sed -e 's/{""},/{"",0},/g' > gperfapphash.cpp" is fine without the error in #295 and #492 .
According to the line 132 to 140 of qucs/qucs-core/src/Makefile.am
BUILT_SOURCES = equation.cpp \
gperfappgen.h \
parse_citi.hpp \
parse_csv.hpp \
parse_dataset.hpp \
parse_mdl.hpp \
parse_netlist.hpp \
parse_touchstone.hpp \
parse_zvr.hpp
does it mean the make process need to check the existence of *hpp files? but I cannot find them in the qucs-core/src directory. When and how are this files generated? I am confusing. Maybe @felix-salfelder  know about it.

felix-salfelder, 2016-05-26:
`BUILT_SOURCES` lists stuff that is built "before everything else" (c.f. the manual for details). its unrelated to creating the rules. remove them from BUILT_SOURCES, and it will likely fail later, with an error much less obvious than

    make[2]: *** No rule to make target parse_citi.hpp', needed byall'. Stop.

the `parse_stuff.hpp` rules are supposed to be there, simply because these are intermediate files required to compile `libqucs.la` from `parse_stuff.ypp` (listed in `libqucs_la_SOURCES`). 

.. we've been there already. see `qucs-core/bootstrap` line 29ff. there, we had blackliststed automake `1.11*`. and we should likely blacklist 1.10* for the very same reason (can anybody check?).

long answer: there *might* be a way to make things work as intended, even with older automake. older automake uses `.h` as the extension for headers related to yacc stuff, and creates rules accordingly. theoretically, this is not a problem. but afair there are places with hardcoded `.hpp` extension in non-automatic files. i couldn't figure it out...

stonecome, 2016-05-28:
hi, @guitorri , @felix-salfelder 
I use automake1.15 to compile qucs-core. It is ok now. :)
Much thanks for your help.

closed 2016-11-12 by ra3xdh