Qucs/qucs #513: qucs crash when editing component properties

kraus1049, 2016-05-25:
### Reproducing way

```
1. Put "dc Voltage source"
2. Right click and choose "Edit Properties"
3. Press Enter key twice
```

qucs will fail into segmentation fault after step 3.

### qucs version

```
$ qucs --version
Qucs 0.0.19 (9d924ca)
```

### Environment

```
＄ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04 LTS"

$ arch
x86_64
```

andresmmera, 2016-05-25:
Hello, 

I've tracked down the seg fault. It seems that this [line](https://github.com/Qucs/qucs/blob/master/qucs/qucs/components/componentdialog.cpp#L723) should be:

  if ( row < prop->rowCount()-1) {

row is an integer ranging from 0 to prop->rowCount()-1, so Qucs tries to access a non-existing element...

prop->setCurrentItem(prop->item(row+1,0));

row+1 > prop->rowCount()-1 => here's the seg fault

Should I open a PR with the patch?

in3otd, 2016-05-25:
> Should I open a PR with the patch?

why shouldn't you? :grin:

andresmmera, 2016-05-25:
just checking :-)

in3otd, 2016-06-18:
Fixed in #514.

Thanks @kraus1049 for reporting the issue!

closed 2016-06-18 by in3otd